guysoft/OctoPi #405: Issue with the prepared PI images

SleepyBrett, 2017-08-26:
It looks like there may be an issue with the automated build. Both 2017-08-19_2017-08-16-octopi-stretch-lite-0.15.zip and 2017-06-25_2017-06-21-octopi-jessie-lite-0.14.0.zip produce the following error when I attempt to restore them onto a 32gb card using Apple PI Baker.

[image]

If I click through the error it claims to have written the file in a little under a minute and it, of course does not boot.

I also made an attempt to write it with dd `sudo dd bs=1m if=2017-08-16-octopi-stretch-lite-0.15.img of=/dev/disk2 conv=sync` also completes in about 20s with the following output:

    dd: /dev/disk2: end of device
    31+0 records in
    30+1 records out
    32100352 bytes transferred in 4.729115 secs (6787814 bytes/sec)

ntoff, 2017-08-26:
Did you try checking the hash of the zip file to make sure it downloaded properly? `2017-08-19_2017-08-16-octopi-stretch-lite-0.15.zip` seems to write to SD fine here (using win32 disk imager), although I do think something is severely broken with the stretch build because a fresh install and boot and neither haproxy, nor webcamd, nor the octoprint service are running for me on my raspberry pi3. But besides that, the image itself doesn't seem to be corrupted.

SleepyBrett, 2017-08-26:
Pretty sure it's just a bad card, headed to the corner now to pick up a new one to test.

SleepyBrett, 2017-08-26:
Yup, my bad, failed card.

closed 2017-08-26 by SleepyBrett