guysoft/OctoPi #408: OctoPi Crashes During Startup with Camera, ODD LOG

AcrimoniousMirth, 2017-09-04:
So every time I start OctoPrint with a camera attached the server crashes and gives me a warning to check the octoprint log.
NOTE: on Pi Zero.
This happens with 2 different Pi Cameras and a Logitech webcam (all the cameras I have on hand). 
It loads fine when nothing attached.
It loads fine when only pi camera cable attached.
It fails when Pi Cam attached or webcam attached

Octoprint log for every failure consistently reads as the last entry the following type of error:

2016-09-04 18:42:11,167 - py.warnings - WARNING - /home/pi/octoprint/local/lib/python2.7/site-package/OctoPrint-1.3.4-py2.7.egg/octoprint/util/__init__.py:591: UnicodeWarning: Unicode unequal comparison failed to convert both arguments to unicode - interpreting them as being unequal
   If value_source != value_target:

Seems a bit odd to cause a server crash but it's the only issue shown in the log file.

guysoft, 2017-09-10:
Are you using a 2 Amp power ssupply with a good USB cable? Might be running outbof power

AcrimoniousMirth, 2017-09-11:
3A and good cable. One of my experiments was to break out the power and one of the GND on the ribbon cable, add in a 3v3 voltage regulator and send power up the broken out power line at 5V. The idea being that that would give a nice, clean power input and come straight from my more powerful 12V supply.
Made no difference

Zippochonda, 2017-09-25:
Having the same problem with a Raspberry Pi Zero W and a PiCam, without Cam the startup is working very good.

guysoft, 2017-09-25:
What version of OctoPi? Some people on stack exchange (can't find the link) complained about not up to date systems

Zippochonda, 2017-09-26:
Version 0.14

mikecy22, 2017-10-07:
A similar problem happend to me. I used a noname web camera and the octopi server showed error. I used to ssh, kill the video process and the restart it with the right resolution and mode for my camera.