guysoft/OctoPi #410: Hostname change not reflected on welcome text

quietbiker, 2017-09-07:
Hi, as I've a few 3D printers, I needed to change the hostname on a new Pi.

The hostname was saved and the Pi rebooted.

The "welcome" text on reboot still refers to "octopi.local" whereas my true hostname is "i2" (so "i2.local")
This is also on the SSH session.

Not a huge problem for me as I know I need to visit i2.local (or via IP), but a minor issue that I thought I would report.

John-Mc, 2017-09-07:
The Hostname is hardcoded into one of the files and is editable (I just don't remember the name or location of the file off the top of my head -- MAYBE `/etc/rc.local` ??

I'm thinking this one was already reported, and that it has been noted for a change to display the actual current hostname in a future version of OctoPi, but I may be mistaken on that.

quietbiker, 2017-09-08:
Thanks John,
it's not really an issue for me, but something I spotted and thought it should be reported for a possible future fix.

On the pi, the hostname is under /etc/hostname
The "welcome text" (part of octopi) needs to read this file rather than assume "octopi" as the hostname.
As I say, I know my hostname, so it's really for new users and something to tidy up.

ntoff, 2017-09-08:
Already reported (and I think fixed?) https://github.com/guysoft/OctoPi/issues/385

guysoft, 2017-09-24:
Indeed, reported and fixed afaik

guysoft, 2017-09-24:
Ah wait.  Its not

GrueMaster, 2017-10-07:
The /etc/rc.local file needs updating as well as the welcome file in /home/pi/scripts.  Also, you should use hostname -f to get the FQDN, instead of appending ".local".  Or check to see if hostname -d is local.  On my router for example, everything is .domain (for now, will change once I get my real firewall back online - CenturyLink grmbl).

guysoft, 2017-10-08:
I'm not finding time to fix this. Just taking to opportunity to mention that pull requests are welcome.

quietbiker, 2017-10-08:
Ok, this is a simple fix (I am a coder, but very limited experience on Linux.

The .local suffix is not really recommended, but is still very common.  Assuming we'll stick with this, I'll sort out a pull request and change both files.

nophead, 2017-10-08:
I can access my machines as octopi.local or octopi.home so I don't think it needs to change does it? octopi.local is the bonjour name and .home is assigned by my router. The router domain is random but .local is part of the octopi configuration.

Stady234, 2018-02-12:
By adding this first section in both the /etc/rc.local and /home/pi/scripts/welcome, the name updates.

- however i go get a " [OK] started Permit User Sessions" entry in between the hostname and IP on the "welcome" screen...
`for name in $(hostname);
do 
echo "    http://$name.local"
done

for ip in $_IP;
do
    echo "    http://$ip"
done

`

commit 98bf2928 (2018-02-13): Fixes #410
closed 2018-02-13 by guysoft