guysoft/OctoPi #411: Cannot log in

dahlheim2, 2017-09-09:
Very simple. Tried three versions including latest stable, and both original RPi and new v3.  Forget networking, that works on ethernet cable. But I cannot log in with username pi password raspberry, either with keyboard or via ssh. Seems like wrong password.

guysoft, 2017-09-09:
What ssh client are you using?

dahlheim2, 2017-09-10:
just what's built into linux (ubuntu mate 17.04).  it's not the ssh client.  even fails with straight console login with usb keyboard and hdmi monitor installed, without ethernet connection.  it worked after the old shell boot hack and changing pi's password that way. weird.

guysoft, 2017-09-10:
In that case you can try ``ssh -v`` to check out verbose output

CedricGoby, 2017-09-30:
I confirm this issue.
I have dowloaded the image from this page : http://octoprint.org/download/
I have checked the MD5Sum and unzip the image and install the contained .img file to an SD card.
I have booted the Pi with the SD card
Then I launched the Terminal (I work with Xubuntu 16.04) and try ``ssh pi@octopi``
I was prompted for the password 
``pi@octopi's password:``
and wrote ``raspberry`` but I got the following message :
``Permission denied, please try again.``
I have also tried by connecting a keyboard and a hdmi screen but the result is the same :
``Permission denied, please try again.``
Help would be appreciated...

guysoft, 2017-09-30:
Guys, I am using the 0.14 image and it works for me. Please use -v for verbose output.

If you are sshing from the pi to itself use pi@localhost

dahlheim2, 2017-09-30:
it happened to me consistently and after redownloading, trying different cards, different writers and anything else i could think of, i just had to hack into the card by booting into single-user mode.  you've got a problem in that package, sir.  at least for some of us.

guysoft, 2017-09-30:
Providing you're not providing the output I asked for. If I don't have access to a machine that produces the bug or you don't give me the outputs  I can't troubleshoot the problemm

dahlheim2, 2017-09-30:
sorry, but all is working well now.  i haven't enough knowledge apparently to understand why it was not working, but above is how i solved it.  i wanted to help this morning so i wrote the same downloaded image i used the other day to a micro-sd card (i don't know if it's one of the few i had already tried), using the same method (dcfldd bs=1M) put it into a new raspberry pi 3b (i had tried a different 3b before, as well as a RPi B+) and booted and, well whaddaya know, i can just log in now.  so i assume the problem is not in your package.  i don't understand.  Thanks anyway for the responses.

closed 2017-09-30 by guysoft