guysoft/OctoPi #413: Unable to Connect to Unsecured Wifi

CoolNamesAllTaken, 2017-09-19:
I've been fiddling around for 5 hours now in an attempt to get my raspberry pi to play nice with my university's wifi, but have had very little luck thus far.  I've previously gotten octoprint to run just fine on my home WPA/WPA2 Secured router, but have had no luck so far with my university wifi, an unsecured WLAN that operates on a MAC whitelist.

I am running OctoPrint version 1.34, Octopi version 0.14.0

Here are some relevant files:
[/boot/octopi-network.txt](https://pastebin.com/bXVuuW43)

- My home setup (which works) is listed under the WPA/WPA2 credentials.  I managed to get away without the use of quotation marks, which was nice.
- The config that does not work is under the open/unsecured networks section.  It is the only uncommented code.  I have taken special care to use quotation marks pasted from an untouched portion of the .txt file to avoid issues with mac quotation marks.  All coding has been done in vim in Terminal anyhow, so hopefully I avoided the TextEdit curly quotation mark fiasco.
- I tried changing `inet manual` to `inet dhcp` for giggles, and I got some indication of DHCPDiscover being activated multiple times, but no IP was received so it went back to sleep.  I read somewhere else on here that things should work on OctoPrint with `inet manual` so I changed it back.

[/boot/octopi-wpa-supplicant.txt](https://pastebin.com/QH4sVJhH)
- This file should be unchanged, other than the country code being changed to 'US' at the top.
- I tried adding a network block for my school network with the essid of the network and key_mgmt=NONE, but that didn't fix things.
- The raspi is definitely seeing all the networks, including the one that I want to connect to.  [These are the results of a WLAN0 scan.
](https://pastebin.com/FjFRVMge)
[iwconfig](https://pastebin.com/077EYTFC)
- When I call `sudo iwconfig wlan0 essid "Stanford Residences"`, then call iwconfig afterwards, the iwconfig is unchanged (still shows as "unassociated").

[ifconfig](https://pastebin.com/K5sh2Yj1)
- My school's network operates on a MAC whitelist.  I have registered the Raspi's wifi MAC address, so it should be on the whitelist (e8:4e:06:2b:8a:90)

[syslog of a clean boot](https://pastebin.com/6vNMkuM0)

It may be worth noting that currently I'm ssh'ed into my raspi from my laptop via an ethernet cable, and using that to run all the terminal commands and edit the config files.  Every time I edit a network config file I call `sudo ifdown wlan0 && sudo ifup wlan0` to relaunch the adapter.

guysoft, 2017-09-19:
Did you try setting:
```
network={
	ssid="Stanford Residences"
	key_mgmt=NONE
}
```
In ``/boot/octopi-wpa-supplicant.txt``?

CoolNamesAllTaken, 2017-09-19:
Yep, I tried that exactly in the supplicant file, with the octopi-network file left as shown in the pastebin.  I have a 3A power adapter supplying the pi with power, so I think things are probably find in that department.  The access point is actually much closer and more powerful at school than at home in this case.

guysoft, 2017-09-19:
If you edited ``octopi-wpa-supplicant.txt``, then you need to comment out the settings in ``octopi-network.txt``, leave octopi-network as is or it will interfere.

CoolNamesAllTaken, 2017-09-19:
I tried commenting out octopi-network.txt and adding
`network={
	ssid="Stanford Residences"
	key_mgmt=NONE
}
 `
to the octopi-wpa-supplicant.txt file, but now I get the following error when I relaunch the network adapter:
`Ignoring unknown interface wlan0=wlan0-octopi`

By the way, thanks for your responsiveness!  You've been very attentive to this issue so far.

CoolNamesAllTaken, 2017-09-19:
Woohoo!  It's connected now, it looks like I'd forgotten to run `sudo ifup wlan0` before rebooting the adapter.  `/sbin/iw wlan0 link` shows a good connection to "Stanford Residences".  Automatic connection seems to work fine during reboot, too.  Thanks for the help!  If this issue is common, a modification to the documentation to suggest this workaround might be helpful.

guysoft, 2017-09-19:
``octopi-wpa-supplicant.txt``  should be the ddefault. Its new so I didn't update the documentation, aand guides are out if date.

closed 2017-09-19 by guysoft