guysoft/OctoPi #416: Info about remove CuraEngine

brunosso, 2017-09-26:
To build the image without CuraEngine the only modification is on the file config?

https://github.com/guysoft/OctoPi/blob/devel/src/modules/octopi/config#L13

change "yes" with "no" ???

thanks

guysoft, 2017-09-26:
yes, that's it. You can also create a config.local with the changes

brunosso, 2017-09-26:
OK thanks

closed 2017-09-26 by brunosso

guysoft, 2017-09-26:
Oh, I might need to fix thia line so the path is in the right place for config.local to workhttps://github.com/guysoft/CustomPiOS/blob/CustomPiOS/src/config#L23

brunosso, 2017-09-26:
> CustomPiOS/src/config#L23

Guy can you tell me why i getting this error on try to build image?

+++ unmount_image /home/pi/OctoPi/OctoPi/src/workspace/mount force
+++ mount_path=/home/pi/OctoPi/OctoPi/src/workspace/mount
+++ force=
+++ '[' 2 -gt 1 ']'
+++ force=force
+++ '[' -n force ']'
++++ sudo lsof /home/pi/OctoPi/OctoPi/src/workspace/mount
++++ awk '{print $2}'
sudo: lsof: command not found
++++ sudo mount
++++ grep /home/pi/OctoPi/OctoPi/src/workspace/mount
++++ awk '{print $1}'
++++ sort -r
++++ awk -F on '{print $2}'
++++ previous_command='unmount_image $BASE_MOUNT_PATH force'
++++ this_command='pushd $BASE_WORKSPACE'
+++ pushd /home/pi/OctoPi/OctoPi/src/workspace
/home/pi/OctoPi/OctoPi/src/workspace /home/pi/OctoPi/OctoPi/src
++++ previous_command='pushd $BASE_WORKSPACE'
++++ this_command='[ -e *.img ]'
+++ '[' -e '*.img' ']'
++++ previous_command='[ -e *.img ]'
++++ this_command='[ ! -f $BASE_ZIP_IMG ]'
+++ '[' '!' -f
/home/pi/OctoPi/OctoPi/src/image/2017-09-07-raspbian-stretch-lite.zip ']'
++++ previous_command='[ ! -f $BASE_ZIP_IMG ]'
++++ this_command='[ "$BASE_ZIP_IMG" == "" ]'
+++ '['
/home/pi/OctoPi/OctoPi/src/image/2017-09-07-raspbian-stretch-lite.zip == ''
']'
++++ previous_command='[ "$BASE_ZIP_IMG" == "" ]'
++++ this_command='unzip $BASE_ZIP_IMG'
+++ unzip
/home/pi/OctoPi/OctoPi/src/image/2017-09-07-raspbian-stretch-lite.zip
Archive:
/home/pi/OctoPi/OctoPi/src/image/2017-09-07-raspbian-stretch-lite.zip
  inflating: 2017-09-07-raspbian-stretch-lite.img   bad CRC bb196135
(should be d1a35037)
+ exit 1
+ exit_code=1
+ '[' 1 -ne 0 ']'
+ exit 1

guysoft, 2017-09-26:
It says ``Archive:
/home/pi/OctoPi/OctoPi/src/image/2017-09-07-raspbian-stretch-lite.zip
  inflating: 2017-09-07-raspbian-stretch-lite.img   bad CRC bb196135``.

So 
1. You didnt fully download the raspbian image, it says the aarchive is broken.
2. Are you building this on a pi?

brunosso, 2017-09-26:
Yes i build in a pi

brunosso, 2017-09-27:
I try to build in a virtualbox Ubuntu system. The image is successfully
created in workspace directory (have the name of raspbian!?!) but how can i
create the zip file?

2017-09-26 21:09 GMT+02:00 Alberto Scrofani <alberto.scrofani@gmail.com>:

> Yes i build in a pi
>
>
>> It says Archive: /home/pi/OctoPi/OctoPi/src/ima
>> ge/2017-09-07-raspbian-stretch-lite.zip inflating:
>> 2017-09-07-raspbian-stretch-lite.img bad CRC bb196135.
>>
>> So
>>
>>    1. You didnt fully download the raspbian image, it says the aarchive
>>    is broken.
>>    2. Are you building this on a pi?

guysoft, 2017-09-27:
If you don't get a zip file it means the build has failed, please provide the ``build.log``, otherwise I can't know what went wrong.

brunosso, 2017-09-27:
https://pastebin.com/GdqagkMg

this is the build.log

I use my git forked from you https://github.com/brunosso/OctoPi

guysoft, 2017-09-27:
My mistake, it looks fine.
In order to get a zip you need to run ``${CUSTOM_PI_OS_PATH}/release`` from the folder `` ``${OCTOPI_PATH}/src`` .
In the vagrant build it does that automatically.

brunosso, 2017-09-27:
So i need to edit the
https://github.com/guysoft/OctoPi/blob/devel/src/build_dist

and add at the end
${CUSTOM_PI_OS_PATH}/release

??

thanks for the help!!!

guysoft, 2017-09-27:
Or just run it