guysoft/OctoPi #417: Problem with generic webcam on 1.3.4

kruuth, 2017-10-04:
I'm trying to use a webcam that I tested with my linux desktop.  It worked fine there but when I attempt to connect it to OctoPI it get nothing.  I tried testing mjpg_streamer from the command line and I get the following:

pi@octopi:~/mjpg-streamer $ ./mjpg_streamer -i "./input_uvc.so -r VGA" -o "./output_http.so -w ./www"
MJPG Streamer Version.: 2.0
 i: Using V4L2 device.: /dev/video0
 i: Desired Resolution: 640 x 480
 i: Frames Per Second.: -1
 i: Format............: JPEG
 i: TV-Norm...........: DEFAULT
libv4l2: error setting pixformat: Device or resource busy
Unable to set format: 1196444237 res: 640x480
 Init v4L2 failed !! exit fatal 
 i: init_VideoIn failed

output of lsusb:

pi@octopi:~/mjpg-streamer $ lsusb
Bus 001 Device 006: ID 1c4f:3000 SiGma Micro Micro USB Web Camera
Bus 001 Device 005: ID 05e3:0608 Genesys Logic, Inc. USB-2.0 4-Port HUB
Bus 001 Device 004: ID 0bda:8172 Realtek Semiconductor Corp. RTL8191SU 802.11n WLAN Adapter
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9512 Standard Microsystems Corp. LAN9500 Ethernet 10/100 Adapter / SMSC9512/9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

Any suggestions of what is wrong?

GrueMaster, 2017-10-08:
Try letting mjpg_streamer figure out the capture resolution on its own.  Remove the '-r VGA' option.

kruuth, 2017-10-13:
I get the same problem:

./mjpg_streamer -i "./input_uvc.so" -o "./output_http.so -w ./www"
MJPG Streamer Version.: 2.0
 i: Using V4L2 device.: /dev/video0
 i: Desired Resolution: 640 x 480
 i: Frames Per Second.: -1
 i: Format............: JPEG
 i: TV-Norm...........: DEFAULT
libv4l2: error setting pixformat: Device or resource busy
Unable to set format: 1196444237 res: 640x480
 Init v4L2 failed !! exit fatal
 i: init_VideoIn failed

kruuth, 2017-10-30:
Ok.  I think that I see the problem.  Despite the warning in the config, I had to remove everything in the .octopi config referencing this and now the cam works.

guysoft, 2018-01-04:
Ok, closing because looks like its solved

closed 2018-01-04 by guysoft

timur-tabi, 2021-11-22:
I'm seeing the same problem:
```
libv4l2: error setting pixformat: Device or resource busy
Unable to set format: 1196444237 res: 1280x720
Init v4L2 failed !! exit fatal
 i: init_VideoIn failed
Done bring up all configured video device
```
My camera appears to work fine, so I don't understand why I'm getting this message.  Also, I don't understand the "fix" either -- what do you mean by "remove everything in the .octopi config referencing this"?

guysoft, 2021-12-15:
@timur-tabi You cant both run your camera and access it from another process. It says its busy.

SokolovJek, 2023-02-06:
This solved my issue. The problem in my case was that I did not correctly specify the input of the camera stream, by default it is /dev/video0, and in my case /dev/video4.
You can check the connection with the command:
 **v4l2-ctl --list-devices**
```
USB 2.0 Camera (usb-fe380000.usb-1):
        /dev/video4
        /dev/video5
```
And then, at startup, you need to explicitly pass the parameter (/dev/video4):
**/usr/local/bin/mjpg_streamer -i "/usr/local/lib/mjpg-streamer/input_uvc.so -d /dev/video4 -r 1920x1080" -o "/usr/local/lib/mjpg-streamer/output_http.so"**