guysoft/OctoPi #418: FileSystem not accessible and not booting on Raspberry Pi 0

Jhyrachy, 2017-10-04:
Hi,
I flashed using Etcher the last img from the octopi site (not nightly, the filename was: octopi-jessie-lite-0.14.0.zip. I also caclulated the MD5 hash and it was ok).

But the 'second partition' is corrupted and not booting (/dev/sda2 in my system, where /dev/sda1 is /boot/) is also not accessible from others system, included linux ones.

guysoft, 2017-10-04:
Are you sure the SD is not corrupted? 0/14 is stable and was installed tens of thousands of times.

Jhyrachy, 2017-10-04:
I'm pretty sure the SD is ok, since I had clean raspbian before, and now I have 0.13 without any problem

guysoft, 2017-10-04:
I have no clue. Can't reproduce

Jhyrachy, 2017-10-04:
ye, had no idea either, but tought it was worth noticing.

Installing 0.13 I solved, so it's no longer a problem for me, but maybe if someone had my same problem

(My board is a raspberry pi 0 v1.2, if it's worth something)

Or maybe is an etcher bug, don't know

Manichee, 2017-11-05:
I had the same issue. Tried win32diskimager and etcher, along with different cards. .14 gave vfs cannot mount boot on startup of pi. Saw this and tried 13 and it worked first try.

guysoft, 2017-11-05:
Did you check the md5? 
You can also try nightly

Manichee, 2017-11-05:
I did check the md5 against the zip and it was good. Is there one for the actual .img file? I will try the nightly just to see if it works later today.

Manichee, 2017-11-06:
Was able to get it to boot using the latest nightly, but my wifi isn't connecting. The same settings worked fine on 13.

guysoft, 2017-11-06:
Try using in the new versions octopi-wpa-supplicant.txt , the interface name might have changed

guysoft, 2018-01-04:
Closing if no reply

closed 2018-02-07 by guysoft