guysoft/OctoPi #419: Issue building armbian with vagrant

gyrex, 2017-10-11:
Hi guysoft,

I'm having issues building an OctoPi armbian variant using vagrant on Ubuntu 16.04. I followed all the directions on the readme and I've attached the vagrant log when I'm trying to build the image.

Details:
Host: Native Ubuntu 16.04.3 x32
vagrant v2.0.0 
Image: Armbian_5.30_Orangepilite_Ubuntu_xenial_default_3.4.113.zip
Image location: OctoPi/src/image-variants

Am I missing something? Thanks in advance for any help.

[OctoPi vagrant errors.txt](https://github.com/guysoft/OctoPi/files/1374040/OctoPi.vagrant.errors.txt)

gyrex, 2017-10-12:
Hi @guysoft - have you got any ideas mate? I really need to build this and any help would be really appreciated.

It's really strange because per the log I posted, the script inflates the armbian image from the src/image-variants directory:

> Archive:  /distro/image-variants/Armbian_5.30_Orangepilite_Ubuntu_xenial_default_3.4.113.zip
> inflating:

bu then it fails looking for an image somewhere else $BASE_IMG_PATH, maybe src/image?:

> +++ BASE_IMG_PATH=
> ++++ previous_command='BASE_IMG_PATH=`ls | grep '\''.img\|.raw'\'' | head -n 1`'
> ++++ this_command='[ ! -f "$BASE_IMG_PATH" ]'
> +++ '[' '!' -f '' ']'
> ++++ previous_command='[ ! -f "$BASE_IMG_PATH" ]'
> ++++ this_command='echo "Error, can'\''t find image path, did you place an image in the image folder?"'

Am I supposed to put something in the src/image directory? To double check, I downloaded the 2017-09-07-raspbian-stretch-lite.zip and put it in there but it still doesn't work. It doesn't seem logical that the script would need a raspbian image though if I'm using an armbian image from the image-varients directory.

I'm a little confused and not sure if I've missed something...

guysoft, 2017-10-12:
When you inflate ``Armbian_5.30_Orangepilite_Ubuntu_xenial_default_3.4.113.zip`` manually, what files do you see inside?

gyrex, 2017-10-12:
Bloody hell, it's always something simple. Thank you for this simple suggestion @guysoft. There was a nested directory in the zip file, I moved the files into root and the build got a bit further but now it's failing when it tries to do an apt-get update and install required packages. It appears that there's no internet connection within the virtual machine. Is this expected?

guysoft, 2017-10-12:
It tends to be a pproblem with no ``/etc/resolv.conf`` and the DNS server failing.
Take a look what is there.
I think rh, variant tries to add ``nameserver 8.8.8.8`` which is the google DNS server.

gyrex, 2017-10-12:
Found the problem, I was blocking the google DNS servers on my router to use some smart dns services - thanks @guysoft. The build finished but when I write the img, there's no FAT partition to specify the network settings. Should there be a FAT partition?

guysoft, 2017-10-12:
Unfortunately armbian only has one partition.  So OctoPi puts the scripts in /boot on the main partition.

TheAssassin, 2017-10-14:
Putting my two cents in: I'd use anything but the Google DNS servers, they're tracking its users and although anyone is free to use it as a distributor of software I wouldn't ship this as a default. There's a few non-tracking alternatives, e.g., `dnscache.berlin.ccc.de` (a.k.a. `213.73.91.35`), which I use as primary DNS on most of my systems and have made some very good experiences with. Please consider using this and not a tracking one.

guysoft, 2017-10-14:
I'm ok with that if it works. PR it :)

TheAssassin, 2017-10-14:
I've "grepped" the source code and can't find that string. As you mentioned, you believed that it's added somewhere. Mind to point me to it? Shall we add it anyway, even if Google is not used at the moment? (I'd avoid that if possible, as the DHCP server will most likely provide a local one that should be used)

guysoft, 2017-10-14:
All armbian stuff is encapsulated in this folder:
https://github.com/guysoft/CustomPiOS/tree/CustomPiOS/src/variants/armbian

It's set up here:

https://github.com/guysoft/CustomPiOS/blob/CustomPiOS/src/variants/armbian/pre_chroot_script

TheAssassin, 2017-10-14:
Ah, that makes sense, I've grepped the OctoPi directory. I'll PR it.

guysoft, 2017-10-20:
Merged

closed 2017-10-20 by guysoft