guysoft/OctoPi #424: cannot ssh into octopi

lesaux, 2017-10-18:
Hello,

On a fresh install of octopi I found that the ssh server had all of it's ssh keys filesizes to 0.

`/etc/ssh/ssh_host_*` files are all at 0 bytes in size.

To see this I pulled out a keyboard and monitor.

To fix it I wiped those files and ran

`sudo /usr/bin/ssh-keygen -A`

On other distros this usually is included in systemctl ssh scripts.

hopes this helps others.

guysoft, 2017-10-19:
What version of OctoPi?

lesaux, 2017-10-19:
This happened on Octopi 0.14. I am convinced that I had deployed this image earlier without facing this problem, so I will try to reproduce, and see if I can figure out how this happened.

guysoft, 2017-10-22:
Ok, I managed to reproduce this

guysoft, 2017-10-22:
Output of ``/var/log/syslog``:
```
Sep  7 15:54:45 fullpageos systemctl[419]: Removed /etc/systemd/system/multi-user.target.wants/regenerate_ssh_host_keys.service.
Sep  7 15:54:45 fullpageos systemd[1]: Reloading.
Sep  7 15:54:46 fullpageos systemd[1]: Started Regenerate SSH host keys.
Sep  7 15:54:46 fullpageos systemd[1]: Starting Turn on SSH if /boot/ssh is present...
Sep  7 15:54:46 fullpageos systemd[1]: Reloading.
Sep  7 15:54:47 fullpageos systemd[1]: Starting OpenBSD Secure Shell server...
Sep  7 15:54:47 fullpageos systemd[1]: ssh.service: Main process exited, code=exited, status=255/n/a
Sep  7 15:54:47 fullpageos systemd[1]: Failed to start OpenBSD Secure Shell server.
Sep  7 15:54:47 fullpageos systemd[1]: ssh.service: Unit entered failed state.
Sep  7 15:54:47 fullpageos systemd[1]: ssh.service: Failed with result 'exit-code'.
Sep  7 15:54:47 fullpageos sh[433]: Job for ssh.service failed because the control process exited with error code.
Sep  7 15:54:47 fullpageos sh[433]: See "systemctl status ssh.service" and "journalctl -xe" for details.
Sep  7 15:54:47 fullpageos sh[433]: invoke-rc.d: initscript ssh, action "start" failed.
Sep  7 15:54:47 fullpageos sh[433]: ● ssh.service - OpenBSD Secure Shell server
Sep  7 15:54:47 fullpageos sh[433]:    Loaded: loaded (/lib/systemd/system/ssh.service; enabled; vendor preset: enabled)
Sep  7 15:54:47 fullpageos sh[433]:    Active: failed (Result: exit-code) since Thu 2017-09-07 15:54:47 UTC; 98ms ago
Sep  7 15:54:47 fullpageos sh[433]:   Process: 454 ExecStart=/usr/sbin/sshd -D $SSHD_OPTS (code=exited, status=255)
Sep  7 15:54:47 fullpageos sh[433]:  Main PID: 454 (code=exited, status=255)
Sep  7 15:54:47 fullpageos sh[433]: Sep 07 15:54:47 fullpageos systemd[1]: Starting OpenBSD Secure Shell server...
Sep  7 15:54:47 fullpageos sh[433]: Sep 07 15:54:47 fullpageos sshd[454]: error: Bind to port 22 on 0.0.0.0 failed: Address already in use.
Sep  7 15:54:47 fullpageos sh[433]: Sep 07 15:54:47 fullpageos sshd[454]: error: Bind to port 22 on :: failed: Address already in use.
Sep  7 15:54:47 fullpageos sh[433]: Sep 07 15:54:47 fullpageos sshd[454]: fatal: Cannot bind any address.
Sep  7 15:54:47 fullpageos sh[433]: Sep 07 15:54:47 fullpageos systemd[1]: ssh.service: Main process exited, code=exited, status=255/n/a
Sep  7 15:54:47 fullpageos sh[433]: Sep 07 15:54:47 fullpageos systemd[1]: Failed to start OpenBSD Secure Shell server.
Sep  7 15:54:47 fullpageos sh[433]: Sep 07 15:54:47 fullpageos systemd[1]: ssh.service: Unit entered failed state.
Sep  7 15:54:47 fullpageos sh[433]: Sep 07 15:54:47 fullpageos systemd[1]: ssh.service: Failed with result 'exit-code'.

```

lesaux, 2017-10-22:
Great!

guysoft, 2017-10-23:
Possible workaround in comment, but would prefer code will go in to Raspbian: https://github.com/RPi-Distro/pi-gen/issues/128#issuecomment-338678821

guysoft, 2017-10-24:
Had a talk with @shift on IRC
Will implement a workaround and also if fails an output of ``journalctl`` so we can see if there is any error worth recording.

lesaux, 2017-10-24:
sounds good! Thanks for your efforts! Octopi is amazing!

guysoft, 2017-10-31:
Fix in next nightly, there is now a log also in ``/var/log/regenerate_ssh_host_keys.log`` if this happens again.

commit dfeb3d1d (2017-10-31): Add workaround and logging to ssh key generation, fixes #424
closed 2017-10-31 by guysoft