guysoft/OctoPi #425: wlan0-octop on octopi-network.txt wrong?

SphaeroX, 2017-10-20:
Hello,

i just wanted to install octopi and noticed that there are problems with the wifi. which is entered in the octopi-network. txt instead of wlan0 wlan0-octopi.

for example 

## WEP secured
#iface wlan0-octopi inet manual
# wireless-essid put SSID here #
# wireless-key "put password here"

with "wlan0-octopi" it definitely doesn't work when I changed that to wlan0 everything went on.

guysoft, 2017-10-20:
Please use octopi-wpa-supplicant.txt and revert the changes you made

closed 2017-10-20 by guysoft

SphaeroX, 2017-10-20:
Okay, thank you very much, but on the homepage startpage it is written in a different way, maybe you should change that. 

https://octopi.octoprint.org/

guysoft, 2017-10-20:
Homepage is wrong then cc @foosel

foosel, 2017-10-20:
Suffix is intentional, see https://github.com/guysoft/OctoPi/commit/9052ed829be91cd04f9b6c4c01d505ffcb7b914c. 

For only one network, `octopi-network.txt` is completely sufficient (also mentioned [here](https://github.com/guysoft/OctoPi/blob/0.14.0/src/filesystem/boot/octopi-network.txt#L22)). I don't know why you want to force the way more complicated to manage `wpa-supplicant.conf` on everyone.

guysoft, 2017-10-20:
Not the best place to bring it up, but i think this
```
network={
  ssid="Your Wifi SSID"
  psk="supersecretwifipassword"
}

```

Is simpler than this:
```
## WPA/WPA2 secured
iface wlan0-octopi inet manual
    wpa-ssid "ssid"
    wpa-psk "pss"
```

foosel, 2017-10-23:
If you only look at WPA2 encryption, then yes.

I'm talking about the other cases. WEP (which sadly still is used), open... or no wifi at all and instead static configuration of the LAN port. All of that (apart from the latter) is also possible via `wpa-supplicant` - but at least right now it is not yet documented at all in there, so for someone who doesn't usually use Linux it's completely obscure how to do anything but WPA2 with `wpa-supplicant`. 

If you want to switch fully to the new file, fine with me, it certainly does have its advantages. But if you don't want to confuse people and leave the majority of our users for who OctoPi is their first real contact with Linux completely in the dark, this new file needs way way more docs and examples first, and the old one a verbal redirect to the new one so that people following some online tutorial not under our control won't run against a wall. Right now, that's not the case, so IMHO it is not suited as the new default file yet. And before you ask, sorry, but I do not currently have the time to take care of that either.