guysoft/OctoPi #427: Update to Octoprint 1.3.5 breaks OctoPi

VanKurt, 2017-10-23:
I'm using OctoPi on raspberry Pi Zero W. I've tried six (!) times to update OctoPi 0.14 to Octoprint 1.3.5, every time the Pi was broken afterwards. E.g. it did not boot, the web server did not start or the Login-button on the OctoPrint home page was not functional.

Of course I also tried the update with a freshly installed, unmodified OctoPi 0.14 image. Also without success.

Is this normal? Is anyone else having these troubles?
Is there any way to fix this? Will there be an OctoPi 0.15 with latest OctoPrint anytime soon?

Currently I'm simply ignoring the update to OctoPrint 1.3.5 to keep my printer working.

Tiefschneegiftler, 2017-10-23:
I also have a problem updating from 1.3.4 to 1.3.5 with a new Raspberry 3 B. Downloaded the version 0.14 of Octopi, got everything up and running. But when prompted with either the notification that the software is to be updated or when updating from the settings menu it fails, resulting in the octopi to restart.

guysoft, 2017-10-23:
cc @foosel

foosel, 2017-10-23:
I'm not aware of any such issues and apart from one (still incomplete) ticket by you @VanKurt I have also not received any reports. Over the course of the release preparations I also always update multiple times from various OctoPi base installs and start versions.

iceblu008, 2017-10-24:
I am having this problem as well whenever I try to login to the octoprint web page the page reloads and nothing happens.

foosel, 2017-10-24:
If you have the TPSmartlinkPlug plugin installed, [uninstall it via SSH](https://github.com/foosel/OctoPrint/wiki/FAQ#i-installed-a-broken-plugin-how-do-i-uninstall-it-again). See jneilliii/OctoPrint-TPLinkSmartplug#19.

Tiefschneegiftler, 2017-10-24:
Hi there, so there is my Log file:

<script src="https://gist.github.com/Tiefschneegiftler/ecdafdbb99685551342063984f8b2f70.js"></script>

Same problem as with the TPSmartlingPlug?

foosel, 2017-10-24:
@Tiefschneegiftler this is not the same issue as the TPLinkSmartplug issue, that's plugin specific. It's however completely unclear as of now if the issue originally reported in this ticket and the TPLinkSmartplug problem are even related. 

In any case, your problem sounds different in that it manifests in not being able to update in the first place. Please stick to foosel/OctoPrint#2158, @guysoft has no control over the update process (that's an OctoPrint feature unrelated to OctoPi), and for now we have to assume that that's where things are going wrong until proven otherwise.

VanKurt, 2017-10-24:
I tried (and failed with) the update from a fresh OctoPi 0.14 image. So no additional plugins were installed.

Will there be an OctoPi 0.15 with the latest OctoPrint on board? That way I would get around the update process ;-)

guysoft, 2017-10-24:
There is a nightly build

foosel, 2017-10-24:
You can also update manually (how is explained in the FAQ) or just cooperate with me and provide all the requested information *on the OctoPrint bug tracker* so that we can figure out what is going wrong for you. Again, updating from stock version 1.3.4 on OctoPi 0.14 was part of the tests I did prior to this release, and also for every single RC before that, multiple times. If there was a general issue, either I or any one of the people who tested and reported back on the RCs over the past nearly three months would have noticed that. That was not the case so it must be something more rare or restricted to your environment and the only way to solve this is to work with me on this.

HerrRausB, 2017-10-27:
same here - just broke two of my PIs... thanks alot for the night shift... patreon cancelled...

foosel, 2017-10-30:
From a fresh OctoPi 0.14 install just now, updating to 1.3.5:

```
2017-10-30 08:16:34,095   /home/pi/oprint/bin/python "/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins/softwareupdate/scripts/update-octoprint.py" --branch= --force=false "/home/pi/OctoPrint" 1.3.5
[...]
2017-10-30 08:17:03,982 > Searching for futures>=3.1.1,<3.2
2017-10-30 08:17:03,983 > Reading https://pypi.python.org/simple/futures/
2017-10-30 08:17:03,983 > Best match: futures 3.1.1
2017-10-30 08:17:03,983 > Downloading https://pypi.python.org/packages/cc/26/b61e3a4eb50653e8a7339d84eeaa46d1e93b92951978873c220ae64d0733/futures-3.1.1.tar.gz#md5=77f261ab86cc78efa2c5fe7be27c3ec8
2017-10-30 08:17:03,984 > Processing futures-3.1.1.tar.gz
2017-10-30 08:17:03,984 > Writing /tmp/easy_install-BXlhp7/futures-3.1.1/setup.cfg
2017-10-30 08:17:03,984 > Running futures-3.1.1/setup.py -q bdist_egg --dist-dir /tmp/easy_install-BXlhp7/futures-3.1.1/egg-dist-tmp-wBInkm
2017-10-30 08:17:03,985 > Adding futures 3.1.1 to easy-install.pth file
2017-10-30 08:17:03,985 > 
2017-10-30 08:17:03,985 > Installed /home/pi/oprint/lib/python2.7/site-packages/futures-3.1.1-py2.7.egg
[...]
2017-10-30 08:17:04,076 > Finished processing dependencies for OctoPrint==1.3.5
```

I don't know what to tell you. From your log it looks like for some reason it can't fetch a necessary dependency from python's own package index. That might be due to a timeout while attempting to read the package index data (seen that in the past due to network issues) or due to pypi having some issues (personally I have never seen that but it could probably happen) or due to some proxy between you and pypi doing weird stuff (I have no idea how your network setup looks like, only you know that).

If OctoPrint doesn't have an unhindered network connection and hence can't fetch from github and pypi, it can't update itself. And not being able to fetch this data is nothing that a user space program or an image like OctoPi for that matter can fix.

You could try if `~/oprint/bin/pip install futures` on its own also fails, I'd expect it to. You could also check if you are able to do a `curl https://pypi.python.org/simple/futures/` produces a HTML list of packages:

```
pi@octopi3:~ $ curl https://pypi.python.org/simple/futures/
<!DOCTYPE html><html><head><title>Links for futures</title></head><body><h1>Links for futures</h1><a href="../../packages/08/80/a9e000de5b7372a514add2ef614a953be8bc3d13c8d6205455739d0fb189/futures-0.2.python3.tar.gz#md5=b6f1d454aa2ba3c870a3743c154b95d5" rel="internal">futures-0.2.python3.tar.gz</a><br/>
<a href="../../packages/09/9a/5cb5882312ae289aadab9dec2bd6557b25c45b79f3de5b400aafe547f3a5/futures-0.1.win32.exe#md5=3d4dbe19986aa9c654a2294cf6280c97" rel="internal">futures-0.1.win32.exe</a><br/>
<a href="../../packages/10/0e/bde74dc8717243adba113ef415b4c4517c532ce18ca5c476f8f147b9f74e/futures-2.1.1.tar.gz#md5=158422d8727f148c439408a8b423f9bb" rel="internal">futures-2.1.1.tar.gz</a><br/>
<a href="../../packages/10/1c/182c9e959ca708ce611a7bf6b0ab6022d4a0e7edad79e5145c220431b028/futures-3.0.0-py2-none-any.whl#md5=d64edfffae0bbda1f28ceef944c1d7cc" rel="internal">futures-3.0.0-py2-none-any.whl</a><br/>
<a href="../../packages/17/47/1382bf17e7faa123464057df0afc69cca9c7eb14e930734ce8b5bdbfad7e/futures-2.0.tar.gz#md5=bcc58e035661b9be325b6d278005c0f2" rel="internal">futures-2.0.tar.gz</a><br/>
<a href="../../packages/18/ab/85f2f673ac828bc3d1c1ceb92c82382cdcbf890046c77cd74edaa63cb7d7/futures-1.0.tar.gz#md5=0171acd9e2486915ccbe3d0e2e92ec3f" rel="internal">futures-1.0.tar.gz</a><br/>
<a href="../../packages/1c/98/bc0938f1c7c46b14dae9b25fc7edd32a7c26ef78b16a8b90835971f2143b/futures-2.1.3.tar.gz#md5=3de68835c8035dff0495cbe7f5c75beb" rel="internal">futures-2.1.3.tar.gz</a><br/>
<a href="../../packages/25/83/113da7c7832a4efa80ec7fd5d25b2ea431d9a847cd78f1fa98ddd99d591a/futures-3.1.0.tar.gz#md5=ab78b3ed854a38b82af19f3d617d29eb" rel="internal">futures-3.1.0.tar.gz</a><br/>
<a href="../../packages/26/7d/a7e1130b63514cd7192f0ba0b53bdda7f96df88641ee0c97bfed8477f324/futures-3.0.3-py2-none-any.whl#md5=347b9df805d31b3edea63731ea79cb54" rel="internal">futures-3.0.3-py2-none-any.whl</a><br/>
<a href="../../packages/2e/65/81cb1de73f001863f90981f4f5a0cea8da08bf7a9ed2a1d812e8f53fe822/futures-0.1.tar.gz#md5=14c2e3af0f90479792555e20061a4532" rel="internal">futures-0.1.tar.gz</a><br/>
<a href="../../packages/32/e3/1de389e9e4372d106b1644b56cc0b94226a3ad637f96f45a8c3b55fc0709/futures-2.1.5-py2.py3-none-any.whl#md5=e6763f4016c0a6e31a0edc15072cc56e" rel="internal">futures-2.1.5-py2.py3-none-any.whl</a><br/>
<a href="../../packages/36/0d/ff8097028893cc2ba0ad6c1a23f8aa5486728f2077f00989733978a4e289/futures-2.1.2.tar.gz#md5=a6fa247e3c5fe3d60d8e12f1b873cc88" rel="internal">futures-2.1.2.tar.gz</a><br/>
<a href="../../packages/38/25/85ec21efa813e4158dcea8bc676d1f81510af0ca3e2ba32ec47247d4857d/futures-2.1.6.tar.gz#md5=cfab9ac3cd55d6c7ddd0546a9f22f453" rel="internal">futures-2.1.6.tar.gz</a><br/>
<a href="../../packages/4b/19/93f2dd20a2e7eb72348c90b697ef7a0bd20168f973fc269f8986b8e49391/futures-2.1.tar.gz#md5=b1b9879b20e1ca45451124428ebd8157" rel="internal">futures-2.1.tar.gz</a><br/>
<a href="../../packages/4c/dc/f9473006d4c9c52d4a4e977173fbcbfb1a8ef3a57e32e885edf994fd4a45/futures-3.0.3.tar.gz#md5=32171f72af7e80c266310794adc4db46" rel="internal">futures-3.0.3.tar.gz</a><br/>
[... cut ...]
<a href="../../packages/ea/c9/35287369718fc05059e7a9d0d73c53745fe981010b4185b3858e7d46eff1/futures-3.0.0.tar.gz#md5=ed8cd52e316ad1ec1ed3fd1f1f622961" rel="internal">futures-3.0.0.tar.gz</a><br/>
<a href="../../packages/f8/e7/fc0fcbeb9193ba2d4de00b065e7fd5aecd0679e93ce95a07322b2b1434f4/futures-3.0.2.tar.gz#md5=42aaf1e4de48d6e871d77dc1f9d96d5a" rel="internal">futures-3.0.2.tar.gz</a><br/>
<a href="../../packages/fd/6f/3ce6a5e9e0754a02ca3f0f6027c73bcc337dbd18517920eea2db0b369ae9/futures-2.1.4.tar.gz#md5=c6116d2d8c3c1a7c386ce196a2c98a4d" rel="internal">futures-2.1.4.tar.gz</a><br/>
```

If not, there's your problem, somewhere between your pi and pypi.

ntoff, 2017-10-30:
Unable to reproduce here either, octopi 0.14 with octoprint 1.3.4 -> 1.3.5 works perfectly fine.

HerrRausB, 2017-10-30:
the curl command produced the list perfectly, but the pip command returned immediately. and with immediately i mean IMMEDIATELY - so no timeout or whatever. heres the error message:

```
pi@fabmini-II ~ $ ~/oprint/bin/pip install futures
Downloading/unpacking futures
  Cannot fetch index base URL http://pypi.python.org/simple/
  Could not find any downloads that satisfy the requirement futures
No distributions at all found for futures
Storing complete log in /home/pi/.pip/pip.log
```
here's the pip log:
[pip.log](https://github.com/guysoft/OctoPi/files/1427676/pip.log)

as for the assumption of any barreer kinda phenomena: update, upgrade, dist-upgrade, git clones in general and all the rest of the partivular manual update process for octoprint worked just fine. so do other downloads from the internet to my LAN. 

as for diving into the problem's solution: i am just an octoprint / octopi user and it worked fine for a long time - i am obviously not the only one experiencing this issue, so maybe the statement "works fine here" wouldn't be too helpful. but i also understand as this is free software, i cant complain about that attitude...

so what - i cancelled the patreon subscription and may get stuck with this...

ntoff, 2017-10-30:
So you've used it for a long time, and been happy with it, but as soon as there's one issue with it, you jump ship and cancel your support?

>so maybe the statement "works fine here" wouldn't be too helpful

But it *is* helpful, the update process works for many/most, the whole "no news is good news" doesn't fly when issue trackers are involved, developers need to know whether this is a wide spread issue affecting everyone, or one limited to only a few people.

Again, this isn't something any of us can reproduce so figuring out why it doesn't work for you might take time:

    pi@octopi:~ $ ~/oprint/bin/pip install futures
    Collecting futures
      Downloading futures-3.1.1-py2-none-any.whl
    Installing collected packages: futures
    Successfully installed futures-3.1.1

As you can see, that command works fine for me, and obviously it worked for @foosel, and many others who tested the various stages of the release candidates before 1.3.5 was finally released. Getting angry and yelling about cancelling your support won't get your issue fixed any faster.

> i cancelled the patreon subscription and may get stuck with this...

Well with that attitude, you WILL get stuck with it. If you feel like you'd be happier with a paid product and don't wish to assist any further in figuring out what went wrong, might I suggest an [AstroBox](https://www.astroprint.com/products/p/astrobox-gateway), or [Repetier Server Pro](https://www.repetier-server.com/).

HerrRausB, 2017-10-30:
To be clear: I cancelled it, because this issue killed two PIs, took me an entire night to get the stuff fixed again eventho i luckily had images, thus made me miss printing a birthday present in time. then i realize, i am not the only one having this problem and all i read is: "works fine here" (and implicitly: "your fault"). this is what then really pisses me off - i have been working in software development and IT consulting for almost 25 years now and this is exactly the attitude i am all allergic against: user comes up with a problem, developer says: works fine on my machine, cant be, must be you. there would have been a little difference in an answer like "oh - interesting, never saw that happening, we take a look".

you see the difference? no? yes? however... 

as for the cancellation: at the moment i start the patreon subscription, this becomes a paid product to me, because that's what a patreon subscription is: paying for a product. i did that, because i was convinced of this stuff and i didn't wanna be that kinda guy, who wants everything for free and then complains anyway. i have a certain experience what it means to maintain such a product, and when i saw that good ppl like gina and tom lost their jobs at that spanish company i immediately subscribed to patreon.

but as i said - it makes it a paid product then. gina is supposed to get paid for the job she does, but then, if it becomes a paid product, i don't expect neither such surprises nor such responses. 

OK... one could argue, that microsoft sells bug collections and calls them products... yeah right... 

and thanks for the recommendations - i am grown up enough to find alternatives or just wait until this problem will be fixed in later releases. but i got the message of being insulted by criticism - could be a milennial prob? ;-)

ntoff, 2017-10-30:
> (and implicitly: "your fault").

Where did anyone say it's your fault? All we're saying is we can't reproduce it, that doesn't make it your fault, if that's the impression you get then that's your own problem with the way you think.

>To be clear: I cancelled it, because this issue killed two PIs

OctoPrint shouldn't be doing anything that can break the underlying OS. The worst that should happen is that OctoPrint itself no longer runs. If it DOES to something terrible that breaks the entire OS, then that's a major bug that needs to be fixed, and the only way it gets fixed is if people who have the bug, co-operate in a civil manner and provide all the information they can, and make themselves available for testing. You say you worked in software dev for 25 years, that's something I'd expect you to know. How are bugs meant to get fixed if the people fixing them can't reproduce them?

And for the record, I'm not a developer, I was just saying that it works here for me, another user of OctoPrint. The only things I've done that might be considered "development" is change a couple of things in its UI, and make a couple of plugins, but I'm 99% just a user of it.

>but i got the message of being insulted by criticism

Your very first words were about cancelling your support. No offers to help, no "ok, I have this issue, what is required of me" just a big old middle finger to everyone involved, and "you broke my stuff, here's some logs, goodbye".

> i am grown up enough to find alternatives

Yours is not the attitude of a "grown up".

ChinookTx, 2017-10-30:
@HerrRausB :  Is your system time set to the right time?  SSL gets confused when the time is not right.  

Also, what happens when you try:

 ~/oprint/bin/pip install --index-url=https://pypi.python.org/simple/ futures

HerrRausB, 2017-10-30:
ChinooTx:

Time is OK:

```
pi@fabmini-II ~ $ date
Mo 30. Okt 18:32:53 CET 2017
```

which is totally fine and precise.

the "~/oprint/bin/pip install --index-url=https://pypi.python.org/simple/ futures" resulted in:

```
pi@fabmini-II ~ $ ~/oprint/bin/pip install --index-url=https://pypi.python.org/simple/ futures
Downloading/unpacking futures
  Downloading futures-3.1.1.tar.gz
  Running setup.py egg_info for package futures

Installing collected packages: futures
  Running setup.py install for futures

Successfully installed futures
Cleaning up...
```

which looks promising - i'm gonna try to do the manual update tomorrow - thanks for the hint!

glennph, 2017-10-30:
I can confirm I'm having the same issue as HerrRausB . Seems like it can't resolve to http://pypi.python.org

ChinookTx, 2017-10-30:
I believe it's related to something old or leftover in your pip config.  It needs to use SSL but doesn't want to for some reason, hence the HTTP Error 403: SSL is required.    Forcing it to https as I suggested hint to this, but I have no idea what could be wrong.  There are similar issues if you search for "HTTP Error 403: SSL is required pip" on google.  Most of them hinting at a version of pip that is too old. :-/

glennph, 2017-10-30:
I tried upgrading(pip and base system) I'm going to just reformat and start over ...not to hard

HerrRausB, 2017-10-31:
at least its now obviously reproducable...

ravelox, 2017-10-31:
I believe it's a mismatch with the version of pip but I haven't tracked down exactly what.

You can run something like:

~/oprint/bin/pip install --index-url=https://pypi.python.org/simple --upgrade pip

to update pip to the latest version and then ~/oprint/bin/pip can be run without the --index-url parameter.

I did this and got to pip version 9.0.1.

HerrRausB, 2017-10-31:
did that pip upgrade and the result here remains the same:

```
Processing dependencies for OctoPrint==1.3.5
Searching for wrapt>=1.10.10,<1.11
Reading http://pypi.python.org/simple/wrapt/
Couldn't find index page for 'wrapt' (maybe misspelled?)
Scanning index of all packages (this may take a while)
Reading http://pypi.python.org/simple/
No local packages or download links found for wrapt>=1.10.10,<1.11
error: Could not find suitable distribution for Requirement.parse('wrapt>=1.10.10,<1.11')
```
maybe i should try that with another complete blank fresh new install :-(

foosel, 2017-11-01:
What Octo**Pi** version is everyone experiencing this running? You can determine this via `cat /etc/octopi_version` (at least if the OctoPi version isn't that old that this file doesn't even exist). Those problems suggest that it's not 0.14 like the initial post referred to.

Also, if it's still failing after attempting to update pip, `~/oprint/bin/pip --version` would be interesting. I'm thinking that the comment above by @ravelox should actually have read `~/oprint/bin/pip install --index-url https://pypi.python.org/simple --upgrade pip`, to force the https scheme, and that Github's formatting swallowed up the "https://" prefix.

Finally, to get things going fast (note that this will not solve the core issue, merely circumvent it), try 

```
cd ~/OctoPrint
~/oprint/bin/pip install --index-url https://pypi.python.org/simple .
```

Alternatively, in order to roll back [see again the FAQ](https://github.com/foosel/OctoPrint/wiki/FAQ#how-can-i-revert-to-an-older-version-of-the-octoprint-installation-on-my-octopi-image) - but I'm not sure if those pip issues won't cause trouble there.

foosel, 2017-11-01:
After flashing an 0.12 image I still thankfully had lying around, I could reproduce this, and it's indeed caused by a change to the python package index that happened a couple days after 1.3.5 was released and is unrelated to the release itself.

To fix on OctoPi 0.12, SSH into your instance, then first do this:

```
~/oprint/bin/pip install --index-url https://pypi.python.org/simple --upgrade pip
printf -- "[easy_install]\nindex-url = https://pypi.python.org/simple\n" > ~/.pydistutils.cfg
```

That updates pip and also tells the also quite old version of Python's distutils on there to use the https URL instead of the non-functional http one.

Then:

```
cd ~/OctoPrint
git pull
~/oprint/bin/python setup.py install
sudo service octoprint restart
```

ravelox, 2017-11-01:
Yes, I can confirm that I was running with Octopi version 0.12.0. 

I definitely had the = sign in the index-url parameter. From my shell history, the command I ran was:

~/oprint/bin/pip install --index-url=https://pypi.python.org/simple --upgrade pip

glennph, 2017-11-01:
Thanks foosel ,
Didn't reformat yet..Works like a charm and already seen the warning about running .12 on the interface.

Lenbok, 2017-11-03:
I am running OctoPi 0.9 and 0.11 on my printers. Will these versions have the same problem?

(Related, is there a recommended procedure for updating these to OctoPi 0.14?)

HerrRausB, 2017-11-04:
experiencing this on as well octopi 0.13 as 0.14 on two different RPI2 and RPI3

guysoft, 2017-11-05:
Ok, also saw there is an announcement on OctoPrint now. Should I close this or does anyone else want to do the honors?

guysoft, 2017-11-06:
OctoPi 0.9? Wha! Thats so retro! Its like from 2014! You should upgrade.
Even Debian dropped support for stuff on that.

Lenbok, 2017-11-06:
@guysoft Yeah, I should upgrade, thus "is there a general upgrade procedure?" :-) Can it be done in place, or does it mean setting up from fresh after backing up the octoprint config.yaml?

HerrRausB, 2017-11-07:
@lenbok dont upgrade it - keep it... some day it will be of huge collector's value ;-)

Lenbok, 2017-11-07:
@HerrRausB right on! I will put it in the cupboard along with my MakerBot cupcake #511 that I put aside a few years ago, thinking that one day it might be valuable like an Apple I computer.

HerrRausB, 2017-11-08:
@Lenbok yeah - there even are some behavioural parallels between Jobs and Pettis as in screwing the customers :-)

Getting off topic I assume...

closed 2017-11-08 by guysoft