guysoft/OctoPi #430: Running mjpg-streamer with no parameters

kruuth, 2017-10-25:
Is there a way to run mjpg-streamer with no parameters?  If I kill the daemonized process and run this:

mjpg_streamer -i "./input_uvc.so" -o "./output_http.so"

I can stream from my webcam.  However, even after getting rid of the options in boot.txt this is still happening.  Do I need to edit the webcam daemon file in root despite it saying not to do that?

ntoff, 2017-10-30:
When you say you removed the options in "boot.txt" (I assume you meant /boot/octopi.txt) did you mean you set it as empty like this `camera_usb_options=""` or you straight up deleted any reference to the options in octopi.txt?

kruuth, 2017-10-30:
Sorry.  I meant that I removed the options in boot.txt but that didn't do anything.  I wound up having to remove the values in the webcamd file under camera_usb_option in order for it to work because no matter what happened it was still using these.