guysoft/OctoPi #431: Build Armbian

Lino77, 2017-11-01:
Failed to build Armbian OctoPi
see build.log
[build.log](https://github.com/guysoft/OctoPi/files/1432751/build.log)
can you help me please

guysoft, 2017-11-01:
Your log cuts int he middle, please provide a complete log.
Last lines are : 
```
+ cp -v -r --preserve=mode,timestamps /filesystem/root/. /
'/filesystem/root/./etc/network/interfaces' -> '/./etc/network/interfaces'
'/filesystem/root/./etc/resolv.conf' -> '/./etc/resolv.conf'
/bin/cp: not writing through dangling symlink '/./etc/resolv.conf'
'/filesystem/root/./etc/wpa_supplicant/wpa_supplicant.conf' -> '/./etc/wpa_supplicant/wpa_supplicant.conf'

```
Your log cuts mid-copy command

Lino77, 2017-11-01:
this was the log  from OctoPi/src/build.log
here is a nwe build.log 
and the console output
[console.txt](https://github.com/guysoft/OctoPi/files/1433822/console.txt)

[build.log](https://github.com/guysoft/OctoPi/files/1433812/build.log)

guysoft, 2017-11-02:
The error is
```
/common.sh: line 106: /bin/cp: cannot execute binary file: Exec format error
```
Build using the vagrant build method [described in the readme](https://github.com/guysoft/OctoPi/blob/devel/README.rst#building-using-vagrant), there is something strange with your environment.

Lino77, 2017-11-02:
I use the vagrant method

guysoft, 2017-11-02:
Then try - ``vagrant destroy ; vagrant up`` 
I had that once in vagrant and that fixed it. Not sure why.

Lino77, 2017-11-02:
+ cp -v -r --preserve=mode,timestamps /filesystem/root/. /
'/filesystem/root/./etc/network/interfaces' -> '/./etc/network/interfaces'
'/filesystem/root/./etc/resolv.conf' -> '/./etc/resolv.conf'
/bin/cp: not writing through dangling symlink '/./etc/resolv.conf'
'/filesystem/root/./etc/wpa_supplicant/wpa_supplicant.conf' -> '/./etc/wpa_supplicant/wpa_supplicant.conf'
+ exit 1
+ exit_code=1
+ '[' 1 -ne 0 ']'
+ exit 1
+ exit_code=1
+ '[' 1 -ne 0 ']'
+ echo 'build script failed'
build script failed
+ exit 1
Connection to 127.0.0.1 closed.
lino@seerver:~/OctoPi/src/vagrant$

Lino77, 2017-11-02:
now stop here

guysoft, 2017-11-02:
Can you paste stderr terminal output? It looks the log is not showing something.

Lino77, 2017-11-02:
[console.txt](https://github.com/guysoft/OctoPi/files/1439693/console.txt)
[build.log](https://github.com/guysoft/OctoPi/files/1439694/build.log)
build.log and terminal output

Lino77, 2017-11-02:
wait a moment now the exec error comes again

Lino77, 2017-11-02:
[build.log](https://github.com/guysoft/OctoPi/files/1439702/build.log)
[console.txt](https://github.com/guysoft/OctoPi/files/1439703/console.txt)
 so please

guysoft, 2017-11-03:
Error is now
```
/bin/cp: not writing through dangling symlink '/./etc/resolv.conf'
```
So there is a symlink issue.
What image are you using?
Did you modify Armbian_5.30_Orangepiplus_Ubuntu_xenial_default_3.4.113.img.zip?

Lino77, 2017-11-03:
No original from here 
https://dl.armbian.com/orangepiplus/

Lino77, 2017-11-03:
Armbian_5.30_Orangepiplus_Debian_jessie_default_3.4.113.img the same error

guysoft, 2017-11-04:
Can you cd to the CusomPIOS repo and run ``git staus``? Also make sure you have the latest with gi``git pull origin CustomPiOS``
Because there is no ``resolv.conf`` in [/src/variants/armbian/filesystem/root/etc](https://github.com/guysoft/CustomPiOS/tree/CustomPiOS/src/variants/armbian/filesystem/root/etc). And your terminal clear shows there is one in your system.

Lino77, 2017-11-04:
there is on in Octopi/src/variants/armbian/filesystem/root/etc

guysoft, 2017-11-04:
Please git pull the latest version of OctoPi, that was removed months ago:
https://github.com/guysoft/OctoPi/tree/devel/src

Lino77, 2017-11-04:
okay sorry i switch now from master to devel branch an now it works sorry

closed 2017-11-04 by Lino77