guysoft/OctoPi #432: unable to connect octopi via ethernet

thakurabhishek795, 2017-11-05:


guysoft, 2017-11-05:
Please provide information
What version of OctoPi?
How are you trying to connect? What address are you using?
What error are you receiving?

thakurabhishek795, 2017-11-05:
I am using latest version of octopi octopi0.4
I'm trying to connect via ethernet  DHCP connection, 
I'm unable to connect ...
 rj45 of raspberry pi is not blinking, unable to establish communication between router and raspberry
so I don't get any ip  for my pi

thakurabhishek795, 2017-11-05:
[image]
this is my router ip setting

thakurabhishek795, 2017-11-05:
As per instruction I haven't changed anything in octopi-network.txt for wired DHCP connection

ChinookTx, 2017-11-05:
Does your Pi boot up?  Do you have a  monitor connected to it to see?  I've had images not boot up in certain cases.
Have you tried another network cable?

thakurabhishek795, 2017-11-05:
How will I know? 
no, I don't have any monitor  to see 
as you said I tried another cable, again it doesn't work
before installing octopi I am using noobs in the same chip

ChinookTx, 2017-11-05:
I don't have a pi with me for now, but you should at least get a link light on your ethernet port.

Without a screen on the Pi, you'll never know if it boots up successfully so you might be chasing the wrong problem.

wiebej007, 2017-11-20:
I have similar issues. With monitor connected it obtain the wrong IP.
Not sure where/how it is getting it.

wiebej007, 2017-11-20:
The IP that it get 's it 169.254.213.111 my home network is 192.168.86.x
I tried rebuilding the SD card but still the same.
This raspberry PI had a previous working image running for over a year.
It got corrupted so I decided to redo with new image. No DHCP getting this IP  169.254.213.111... makes no sense.

guysoft, 2017-11-21:
169.254.213.111 is a self aassigned ip.

169.254.0.0/16 - This is the "link local" block. It is allocated for
communication between hosts on a single link. Hosts obtain these
addresses by auto-configuration, such as when a DHCP server may not
be found.

wiebej007, 2017-11-21:
Thanks for the feedback.
I figured as mush that it may be self assigned.
What I can't figure out is why it can'd find the DHCP server.
I have another raspberry pi with Raspbian distro. 
Using the same cat5 it has no problems with getting IP using DHCP without any additional configurations. 
Also this particular pi was running a previous version of OctpPI using same physical networking setup. As I mentioned it got corrupted so I decided to just upload new version/image.

ChinookTx, 2017-11-21:
Is it possible whatever "corrupted" your installation also damaged the network interface on the Pi?  I'd try a static IP but if that doesn't work, your Pi might be hosed?

wiebej007, 2017-11-22:
I figured it after trying with a static IP. 
My router setup was bad.
In fact I had corrupted my previous OctoPi image while making some network changes and powering off pi during booting. 

Thanks for all that commented.

guysoft, 2017-11-22:
Ok, fold, closing of no reply from original poster.

closed 2018-02-07 by guysoft

srujanraj, 2018-06-12:
Hi all,
I am using Octopi on CURA software for CR10 3Dprinter. Due to some problem in the connection i had rewritten the image file on to the sd card and inserted it to the board.  Now i am unable to establish the connection when i am trying to open with putty. before this i was able to open configuration in putty but now i have an error "network error: connection failed". Can someone help me on this?

John-Mc, 2018-06-12:
@srujanraj - This sounds more like a set-up probem than a bug. Since this site is dedicated to bug reports and developement of OctoPi, you might want to try posting your question on the support forum for OctoPrint/OctoPi: [](https://discourse.octoprint.org/). I particular, if you are trying to connect via WiFi, I suggest visiting this FAQ on that site: [Getting your WiFi Connection Working](https://discourse.octoprint.org/t/getting-your-wifi-connection-working/184)

guysoft, 2018-06-12:
@John-Mc  You are quite right, but it was posted before the community forum opened so I gave some way.
But yes @srujanraj , better check the forum now.

John-Mc, 2018-06-12:
@guysoft  - I responded to srujanraj because he had just posted here today (4 months after you had closed this topic). Just thought I'd try to save you the time of redirecting.

srujanraj, 2018-06-15:
Guys sorry for late reply. @John-Mc Thank you for the reply and any how the problem is solved. @guysoft i am sorry that i was not aware of all these things coz i am new to this. Any ways nice interacting with you guys and thank you