guysoft/OctoPi #433: WiFi of OctoPi doesn't work anymore after I connect with my mobile phone

cyberius0, 2017-11-07:
Hello,
I have a very strange issue, I had it with the Debian Jessie release and I have it now with the latest nightly (Debian Stretch). I updated and upgraded OctoPi (Raspberry Pi 3) to the latest version via command line.

Often after I connect with my mobile phone (Galaxy S7, Firefox Browser) to http://octopi the GUI opens, but it doesn't show any information like connection status or temperature and after some seconds I get a "can't connect" error. After this I also cannot connect anymore via my desktop PC or via SSH; the connection is refused. Also in FritzBox (my router) the Raspberry seems offline. This mostly happens when Octopi is printing currently, is it is idle, I can connect normally.

But Octopi continues to print, so the print server is still working (!). I guess the WiFi connection breaks. I have to reboot the Raspberry then everything works again. I don't have this issue when I connect via my desktop PC (also Firefox).

I know this sounds strange... Please tell me what LOG files I should provide to narrow the problem down. Thank you for your help.

DerFetzer, 2017-11-12:
Hi,
I have the exact same issue! (OctoPi 0.14 fresh install, Galaxy S5, LineageOS, Firefox)
Do not hesitate to contact me, if you need additional information. Thanks for your effort.

guysoft, 2017-11-12:
@foosel? Any difficult behavior in the server ofr mobile?

foosel, 2017-11-13:
No, none at all. And if the WiFi goes down, it's not anything that OctoPrint could be doing to begin with, it doesn't control the underlying network connections in any way at all as you know.

I'd check the syslog, see if any errors are logged there when it happens. But that's about the only idea I have.

DerFetzer, 2017-11-18:
The problem seems to be the same as in [this](https://github.com/raspberrypi/linux/issues/1342) issue in the kernel repository.

timotl, 2017-12-09:
I was having the same issue with my Pi 3 only when connecting from my phone.
Updating the Pi firmware fixed it for me: sudo rpi-update
Hope it helps.

tedder, 2018-01-04:
Try accessing it by IP address. Android doesn't support avahi/zeroconf/bonjour/.local unless an application developer specifically includes it.

yellowfeeva, 2018-01-10:
Same issue with mine. Access it via phone and the wifi connection drops.

I've updated firmware on my Pi3, tried creating a cronjob auto-restart of wlan0 with no luck.

John-Mc, 2018-01-10:
The fact that it happens when printing, but not when printer is idle makes me wonder if it is related to power supply? One more thing loading it down is enough to push it over the edge? Though I'm not sure why the phone would push it over the edge, and not the PC.

timotl, 2018-01-13:
Turns out that updating the Pi firmware didn't actually fix the problem. 
I tried the  Stretch nightly build from 12-23 and haven't been able to knock it offline for a few days now.

nikhiljha, 2018-02-19:
Am having the same problem. OctoPi was working great for a while until I connected with my phone - now it dies every hour or so.

Since this seems to be fixed on newer raspbian, can the OctoPi image be updated? @guysoft

guysoft, 2018-02-19:
@nikhiljha OctoPi has nightly builds. You can access their page from the README.rst

John-Mc, 2018-02-19:
The OctoPi 0.15 is based on Raspian-Stretch, so if it was Raspian-related, and the newer versions fixed it, you should be OK.

Having said that, I've used my iPhone to check on OctoPrint occasionally, and never had a problem.

guysoft, 2019-07-02:
Closing due to no reply of OP

closed 2019-07-02 by guysoft