guysoft/OctoPi #434: Raspberry pi camera not working

ChemicalXandco, 2017-11-09:
This is on the [raspberry pi 1 B](https://socialcompare.com/en/review/raspberry-pi-model-b) with the [v2 official pi camera](https://www.amazon.co.uk/Raspberry-Pi-Camera-Module-Filter/dp/B01ER4FA9U/ref=sr_1_3?ie=UTF8&qid=1510213440&sr=8-3&keywords=raspberry+pi+camera+v2). I was told that it was plug'n'play and you just popped it into the CSI socket between the ethernet and HDMI with the metal contacts facing the HDMI port then it would work. But it doesn't. In the octoprint web UI where the webcam should be it just says "webcam not available.I used a voltmeter and touched it to the contacts on the pi and camera board. The connection was strong so it's not that the ribbon cable is slightly unplugged. I popped a sd card with plain raspbian onto it and ran `raspistill -o image.jpg` and the camera took a beautiful image of my foot. This tells me that it's a problem with octoprint. I tried running the same command on octoprint via ssh. It just returns 'camera is not detected' The camera is plugged in exactly how it was when I tested it on plain raspbian. No plugins were installed and I have tried it on the latest version of octopi and both 1.3.4 and 1.3.5 versions of octoprint. I appreciate your help and I am really tearing my hair out.

guysoft, 2017-11-09:
You are mixing OctoPi and OctoPrint version - what version of OctoPi?

ChemicalXandco, 2017-11-09:
Ok then.
Octopi is latest (0.14.0)
I have tried both 1.3.4 and 1.3.5 versions of octoprint.

ChemicalXandco, 2017-11-11:
Is anyone able to reproduce this? Please, can someone tell me how I can get the webcam stream running? I've had an idea: What if I use the same version of octoprint as in [This Guide](https://youtu.be/MwsxO3ksxm4?t=3m38s) (0.12.0) I'll try that and I'll report back if it works or not. BTW if anyone else wants to download an older version try downloading it from [Here](http://docstech.net/OctoPiMirror/nightly/)

ChemicalXandco, 2017-11-12:
That didn't work...

eyal0, 2017-11-29:
Maybe it's because of the -f option?

I have this issue, too, and I was able to solve it by adding the USB id of my webcam to /boot/octopi.txt in the list of `additional_brokenfps_usb_devices`.

ChemicalXandco, 2017-12-02:
when I ran `lsusb` I got

```
Bus 001 Device 004: ID 148f:5370 Ralink Technology, Corp. RT5370 Wireless Adapter                                                                                                                                                            
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast                                                                                                                                                              Ethernet Adapter
Bus 001 Device 002: ID 0424:9512 Standard Microsystems Corp. LAN9500 Ethernet 10                                                                                                                                                             /100 Adapter / SMSC9512/9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

No sign of raspberry pi camera. Is it because it is not usb?

eyal0, 2017-12-02:
That's not a USB camera, perhaps.  Maybe just use a regular USB web cam?

ChemicalXandco, 2017-12-03:
I could: but it's not preferable as there are 2 usb, so one for 3d printer and one for wifi and no more left for camera

ChemicalXandco, 2017-12-03:
I moved the raspberry pi to a different location, redownloaded octopi and reflashed it and it worked! I don't know what the problem was, but in the end I don't think it was octopi or octoprint. The camera stream from the raspberry pi camera is all nice save for a little delay but for monitoring prints its fine. Sorry for any bother I caused.

ChemicalXandco, 2018-01-21:
Please see my 5th post on [here](https://forum.astroprint.com/t/raspberry-pi-camera-not-working/2691/21)
Basically, sometimes the ic and camera module starts at the right time and that was what happened just there. It appears to be a firmware issue with Arch Linux.

guysoft, 2018-01-21:
Hey,
OctoPi runs on Raspbian and not Arch linux, now you are also mixing linux distributions.
Moreover you reported this issue was solved. I don't understand what you are doing and how its connected to OctoPi.
If you are using astroprint they have their own image, which is not OctoPi-based.

ChemicalXandco, 2018-01-21:
[image]
As you can see, the webcam stream is working and it wasn't before I did this:

1. Install octoprint as you would normally
2. Run 
    `sudo apt-get update`
    `sudo apt-get install git-core`
    `cd`
    `git clone git://git.drogon.net/wiringPi`
    `cd ~/wiringPi`
    `git pull origin`
    `cd ~/wiringPi`
    `./build`
    That will get WiringPi good to go
3. Now run 
    `nano camerafix.sh`
    Then paste

    #!/bin/bash
    gpio -g mode 0 in
    gpio -g mode 1 in
    gpio -g mode 28 alt0
    gpio -g mode 29 alt0

    or (I havn't tested this one yet)
    
    #!/bin/bash
    gpio -g mode 0 in
    gpio -g mode 1 in

    Then press Contol+X then Y then Enter
4. Now you run `chmod +x camerafix.sh` and then go to control and you will see the stream is working

5. I think you need to do this every time you start it up so now to make it run at startup, run `sudo nano /etc/rc.local` and at the line just above `exit 0` put the command from number 4 there. 
    Then press Contol+X then Y then Enter

Sorry for the confusion, but I thought it had fixed itself.

guysoft, 2018-02-07:
Dude. you are MIXING between OctoPi and OctoPrint, I am closing this.
OctoPi is distibuton that comes already with OctoPrint. You wont be installing it if you were using OctoPi.

closed 2018-02-07 by guysoft