guysoft/OctoPi #436: PARTUUID incorrect in 0.14.0 release

mjeveritt, 2017-11-11:
It would appear that in the 0.14.0 release, the PARTUUID parameter given in the boot partition cmdline.txt file, and the image /etc/fstab are both incorrect (starting with 0x3nnnn, where mine were 0x5nnnn). This causes the image to hang after loading the kernel (I managed to debug this by enabling the serial console and removing the 'quiet' option from the kernel command-line. I'm 98% sure this is not a configuration error my end, and I will be investigating the build process to see if there's a patch needed to correct this.

guysoft, 2017-11-13:
So how is it running on so many machines?
I mean I have it running here.

mjeveritt, 2017-11-14:
Perhaps it gets changed when I increase the size of the rootfs partition after I write the image - I don't have much experience of PARTUUID's but I know it doesn't happen with UUIDs. I will do some further testing and update the ticket!

mjeveritt, 2017-11-14:
OK, so 5 minutes of research, and it's entirely my fault for resizing the file-system. I think Raspbian uses a particular method which presumably avoids this particular gotcha, so perhaps just a one-liner warning on the Installation notes would catch any other person daft enough to do it manually before booting the card!!
Closing issue as a Red Herring! :smiley:

closed 2017-11-14 by mjeveritt