guysoft/OctoPi #437: Rebuild Pi kernel to enable UUID on kernel cmdline and in /etc/fstab

mjeveritt, 2017-11-11:
Currently the Raspberry Pi kernels only support PARTUUID for partition identification. It would be helpful to use the full UUID for this, as being more distinct, but this requires building a small initrd to permit the appropriate /dev/by-XXX to be populated by mdev/etc. The kernel configuration options for this is documented in the R-Pi kernel repository in [issue #392](https://github.com/raspberrypi/linux/issues/392).

guysoft, 2017-11-12:
Adding this to the build would make it take about 12 hours (time it takes to build RealtimePi).However making a variant should be easy with the kernel module. Or a base-image with a custom kernel and use that.

I'm not sure I fully understand why you would want this feature in OctoPi, how does it benefit the used?

mjeveritt, 2017-11-14:
Are you building on host - I am fairly good at cross-building kernels on x86_64 hardware, so it won't be onerous for me! I'll see if I can work out the config, and give your scripts a whirl to see if I can inject a new kernel into the distro.

The reason for the feature is that UUIDs are fairly universal and fairly stable as partition identifiers, and whilst they do require a bit of extra trickery to 'work' I think the benefit does outweigh the cost.

guysoft, 2017-11-15:
What kind of impact this has to the user? I am not sure if changing the kernel is worth it, because I can't think of a scenario where this would change anything to the end user. I also asked this in the last comment. Please explain if you do know of a scenario.

guysoft, 2017-12-26:
Hey, 
There is now a [kernel module in CustomPiOS](https://github.com/guysoft/CustomPiOS/wiki/Modules#kernel). This lets you build a custom kernel within CustomPiOS. However it takes a good 12 hours. You could easily make your own variant.
Also, once you have a base image you can build OctoPi on top of it.

closed 2018-01-04 by guysoft