guysoft/OctoPi #441: Add notes to editable files about NOT using windows "wordpad"

ntoff, 2017-11-26:
Twice now I've seen an issue on the G+ community where someone has used windows' "Wordpad" to edit octopi.txt and it's broken it. Perhaps the same warning that applies to Text Edit's evil EVIL "smart quotes" needs to be extended to include not using Wordpad *at all*, and also a warning note put into the octopi.txt file about only using recommended text editors like sublime or notepad++ if you're on windows.

guysoft, 2017-11-27:
We have [this warning](https://github.com/guysoft/CustomPiOS/blob/CustomPiOS/src/modules/network/filesystem/boot/custompios-network.txt#L13), If you have an idea how to extend this message you are welcome to pull request it.
It has instructions how to not get around this issue for mac users. I don't recommend people download something new. I am pretty sure windows have a way to edit plaintext files.

ntoff, 2017-11-27:
That warning doesn't exist in octopi.txt (no warning does), and warnings for mac users don't help anyone using windows, and using a different piece of software.

Windows wordpad doesn't just break the quotes, it also breaks the line endings, it totally screws up the file in more ways than textedit does. 

Windows uses a completely different line ending format https://en.wikipedia.org/wiki/Newline so users need to edit the files using a file editor that is aware of unix style line endings.

https://plus.google.com/+morherman/posts/RnevFz8QHKC

This is the second post I've seen where someone has used wordpad to edit octopi.txt, completely breaking it. Notepad++ is a trusted and well known text editor (and free) so if you don't suggest they download anything new, a warning should still be in place that using wordpad will completely obliterate the config.

guysoft, 2017-11-27:
I said there is "this warning", and if you want to PR one for windows users you are welcome.

nophead, 2017-11-27:
How about a startup script that detects bad line endings and quotes and
simply fixes them? Can't be that difficult and much better than warnings
that get ignored.

guysoft, 2017-11-27:
I'd accept a PR like that too

ntoff, 2017-11-27:
warnings in the text file I can handle, I don't know how to do the format checking script though.

ntoff, 2017-12-05:
I started on the extra warnings but as nophead said, they'd probably get ignored anyway, and the configureation files are cluttered with comments enough as it is. 

Either centralised documentation with details on editing (like a `README BEFORE EDITING` file in /boot), or just the checking script would probably be neater. Unfortunately users tend not to read documentation, and I don't know how to do the checking script.

guysoft, 2017-12-26:
Indeed, so should I close this? Or is there some action that can be done?

ntoff, 2017-12-26:
It's up to you, you can leave it open if you want, as a reminder something might need doing some day, like some sort of checking script, or just close it if you don't plan on doing anything about it (like I said I don't know how to do the script, otherwise I'd do a PR).

John-Mc, 2018-02-05:
It would also be helpful if the Getting Started With OctoPi instructions at https://octoprint.org/download/ made some mention of the fact that there are issues with the two programs that people new to this are probably most likely to use (Windows Wordpad, Apple's TextEdit. Not sure if Windows Notepad also has problems?)

The only mention I can find of Notepad problems is a brief mention at about 4:25 in Tom's "Getting started..." video on that download page. That is so quick it's easy to miss, and a lot of people will work from the text description, rather than the video.

guysoft, 2018-02-06:
We always get the feeling people never read, but @foosel , you can probably add that to the page. Might get a few people in the right direction.

foosel, 2018-02-07:
[x] done

John-Mc, 2018-02-07:
Cool!  I use BBEdit on the Mac. The free version has all the features I need (all of the features of the now-discontinued TextWrangler are in the free version of BBEdit)