guysoft/OctoPi #444: Consider adding slic3r to the image

eyal0, 2017-12-01:
OctoPi already includes a cura plugin and cura binary.  What about adding a slic3r binary?  Do you think that some people would want it?  Slic3r is actively developed and very customizable.  Some people say that it's better than cura, though it's slower.

I've worked on the install instructions for RPi.  if I understand correctly, to add Slic3r, the build would need to be part of [this](https://github.com/guysoft/OctoPi/blob/dfeb3d1d744b37f5196bc218d4373f4bb3f067cb/src/modules/octopi/start_chroot_script#L92)?

it would make the image larger, however, and OctoPrint-Slic3r isn't bundled with OctoPrint by default so maybe it wouldn't be used much?

Anyway, something to consider.  If you want Slic3r as part of OctoPi, let me know and I'll try to help.

guysoft, 2017-12-02:
We could add  Slic3r, if it doesn't make the image considerably bigger I'd consider it. At least adding an installation script could totally work.
[These are the instructions how to add it](https://github.com/OctoPrint/OctoPrint-Slic3r/wiki/How-to-install-Slic3r-on-RPi). I'm pretty busy, however pull requests are welcome.

eyal0, 2017-12-02:
I know those instructions, I wrote them!

You're right, another option to to provide a script. That seems like a good
choice considering that the Slic3r plugin isn't installed by default.

Assuming that we provide a script, how will users find out about it? It
seems a shame to make users have to ssh/telnet to install. Maybe a button
from the OctoPrint-Slic3r plugin?

One advantage of providing slic3r in the image is that the build is very,
very slow. It can take 2 hours. Maybe OctoPrint-Slic3r could provide some
sort of zip file or image that includes just Slic3r and then users could
use that...?

I'm open to ideas.

Eyal

<https://github.com/guysoft/OctoPi/issues/444#issuecomment-348685868>, or mute
the thread

Penfold81, 2017-12-02:
Is this really needed? Since slic3r has the ability built in to export direct to octoprint already? If people want slic3r then get slic3r. It’s not like it costs anything?

eyal0, 2017-12-02:
@Penfold81 Some OctoPi installations might be used by any people and requiring them all to install Slic3r isn't convenient.  Having it installed on the OctoPi can service all users.

For the same reason that it makes sense to have Cura installed by default, so might it make sense to have Slic3r installed.

guysoft, 2017-12-03:
I wouldn't go for pre-built binaries unless you maintain a deb package (which isn't much fun).
I'm willing to consider adding ot to ththe build if its not going to bloat up the image.

ChristiaanDev, 2017-12-04:
I support this, i and i suppose quite a lot of people prefer slic3r over cura. Some kind of script would make sense in my opinion.

eyal0, 2017-12-04:
Maybe two scripts, one for alexrj's original slic3r version and one for
Prusa's version, when check is also well-developed.

Will users really want to log in to their raspberry pi and run a script?

ChristiaanDev, 2017-12-05:
>Maybe two scripts, one for alexrj's original slic3r version and one for Prusa's version, when check is also well-developed.

Yeah i can't agree more, i think more choices is just going to attract more people to octopi.

>Will users really want to log in to their raspberry pi and run a script?

I think they would if they want slic3r, as long as it's not a process that takes multiple hours.
I don't think they'd mind the extra few kilobytes for extra choices (in case they don't need slic3r) in octopi either.

ismangil, 2017-12-09:
@eyal0 for the first release, logging in via ssh to add Slic3r capability should be fine. OctoPi already enable SSH by default anyway, which is good. So I support some kind of installation script. But also if it can ship builtin OctoPi even better!

eyal0, 2017-12-09:
@ismangil But which slic3r version to include?  alexrj or Prusa?  I use Prusa's because I have a Prusa printer but both are actively developed.  Either make sense.

ismangil, 2017-12-09:
If I had to choose one, alexrj. Good enough for most users like Cura 15. Advanced users will slice on PCs anyway.

ChristiaanDev, 2017-12-10:
> But which slic3r version to include? alexrj or Prusa? I use Prusa's because I have a Prusa printer but both are actively developed. Either make sense.

I'd prefer Prusa's one as well, however i'd like it if we could just choose within the script/dialog. Would make most sense to me.

ismangil, 2018-01-04:
@eyal0 After going through the Slic3r/PE install instructions (you're right, it took hours), I ended up not trusting it, because there's a broken STL file that prints fine on the Cura plugin, yet  prints out wrong on both Slic3r. Yes, of course any broken file should be fixed in netfabb or whatever, but the point is for some broken files Cura 15.04 can autorepair it. It just works.

eyal0, 2018-01-10:
Is it broken also when using slic3r on your desktop?  If so, it might just be slic3r's problem.  Slic3r isn't perfect.

da11an, 2019-04-03:
I would love to be able to choose a Slic3r instead of Cura image. And the choice between PE vs Alexrj would also be really nice. I just installed it myself (which took a while) but I think I messed up my networking settings in the process. It would be nice to have a ready made image in case you decide to wipe it and start over.

eyal0, 2019-04-03:
What's the best way to do this?  Including the script into OctoPi would be one way, but the script downloads the needed files and those files might change or be inaccessible.

How does OctoPi handle other things, like OctoPrint and cura?  Are they included as binaries without requiring download or are they downloaded during the install process?

John-Mc, 2019-04-03:
Cura is a plugin for OctoPrint. Up to now, it has been bundled with OctoPrint. I believe I read that Gina is planning on unbundling it (it will still be available as a user installed plugin). Because of how well OctoPrint interfaces with many modern desktop slicers, few people are actually using the outdated Cura plugin anyway. I'm doubting she is going to reverse herself and start bundling a slicer again. I'm betting she'll leave this to the plugin authors out there.

guysoft, 2019-04-03:
Technically its also my decision.
I think it might be a good idea to ship slicer installers. So your OctoPi
would ship with a script that would download and install everything.
That way you both don't increase the footprint of the image, but also make
it as easy as possible to install.

John-Mc, 2019-04-03:
That would be a great solution, @guysoft . I'm still hoping that someday someone does an updated Cura plugin for OctoPrint

eyal0, 2019-04-03:
> I think it might be a good idea to ship slicer installers.

This seems like a good idea to me.  The script that I wrote isn't very stable, I worry, because it connects to servers that might or might not exist.

I do not have an RPi to test.  If someone else wants to turn https://github.com/OctoPrint/OctoPrint-Slic3r/wiki/How-to-install-Slic3r-on-RPi into scripts that will install slic3r, I can try to help.  Does someone want to do it?

guysoft, 2019-04-03:
We could ship your script in current nightly, perhaps add a basic internet
check first for those servers.

eyal0, 2019-04-04:
We need someone to test it.  I have no idea if it even works today because I haven't used it in a long time.

I suggest two scripts, one for Slic3r and one for Slic3r PE.  Each should install to completely different directories so that people could potentially install both if they wanted.  They should be pinned to a specific, known-working version to prevent errors.  It should be easy to reclaim the disk space after so I suggest that all installing is done in subdirectories under a single directory, that way a single command can clean it all up.

Who will do this and test this?  I don't have an RPi.

guysoft, 2019-04-05:
@eyal0 What I need to start, is a script I can push to ``/home/pi/scripts/install_slic3r.sh`` and also perhaps have a value to pick up other remotes (like PE).

I see instructions but no script.

If I paste it line by line I get:
https://gist.github.com/guysoft/6233e9911d6bd55e08e898732b29479a

VuiMuich, 2019-04-05:
I have a pi0w and micro as cards in the mail. As soon as they arrive I might give your script a shot.

eyal0, 2019-04-05:
> If I paste it line by line I get:
> https://gist.github.com/guysoft/6233e9911d6bd55e08e898732b29479a

That script looks right but something is required after the second line to make it work.  I have used a reboot but I think that there is another command that does not require a reboot that can be used.

According to [this](https://www.bitpi.co/2015/02/11/how-to-change-raspberry-pis-swapfile-size-on-rasbian/):

```
sudo /etc/init.d/dphys-swapfile stop
sudo /etc/init.d/dphys-swapfile start
```

Probably `restart` would do the same.  So that line of the script needs adjusting.

Lines 12-77 are not needed for alexrj version.  Those are just for slic3r Prusa edition (slic3r PE).  The build can be a little faster without those lines.

eyal0, 2019-04-05:
> I have a pi0w and micro sd cards in the mail. As soon as they arrive I might give your script a shot.

That would be good.  Ideally you could take a fresh install of OctoPi and run the script on it.  We should try to get the script to the point where it works smoothly on all devices.  Both alexrj and slic3r PE.

guysoft, 2019-04-11:
I really don't have time to test these scripts, would appreciate is someone supplies something that was actually executed, and actually gave an output.

VuiMuich, 2019-04-28:
> That would be good. Ideally you could take a fresh install of OctoPi and run the script on it. We should try to get the script to the point where it works smoothly on all devices. Both alexrj and [...]

in the last two weeks I did some testing on my pi0w.
First I have to say, it is no fun to compile Slic3r on this thingy, so maybe the best would be to have some binary repository of some sort.
Build times for the alexrj version in master branch were around 24h for me. The PE fail roughly at 50% +-5 the longest build I let run was 72h, then canceled and tried to change something. On this one I tried master and 1.42.0_beta2 and 1.41.3 (which failed to clone from Github for some reason I have not researched for). Though the "master" seamed to fail during a general error in the branch, since there seams to be some work to be undergoing on the file structure of the project itself.

Maybe Prusa could provide ARM binaries, possibly even stripped down to only the libs and cli-tools needed for octoprint/octopi.

guysoft, 2019-04-28:
@VuiMuich I doub't pursa would provide arm builds.

Again, see my message above, , would appreciate is someone supplies something that was actually executed, and actually gave an output. 
If you can provide a build script, I can hold builds and binaries.

VuiMuich, 2019-05-01:
I just learned that the Slic3r PE build is failing because the libboost in stretch is only 1.62 but needs to be at least 1.64.
Right now I am trying if upgrading to buster (where libboost is 1.68 I believe) would help.

I will try to put a script together from eyal0's guide for the alexrj version.

guysoft, 2020-06-10:
Hey, so I am considering adding this, mainly because the current cura is so old it won't compile on 64bit, and frankly I don't know who is using it. And I find that slic3r seems to be a better fit. And now with stronger Pis out there slicing on the device makes more sense in some cases.

guysoft, 2020-06-10:
@eyal0 is your script up to date? It needs to be converted to a module in order to be added in OctoPi.

RyanMorash, 2020-06-14:
I just the other week went through the process of building Slic3r on my Pi 4. The guide at https://github.com/OctoPrint/OctoPrint-Slic3r/wiki/How-to-install-Slic3r-on-RPi did not work for me. What I ended up doing was building the dependencies using the makefile included with PrusaSlicer (https://github.com/prusa3d/PrusaSlicer/blob/master/doc/How%20to%20build%20-%20Linux%20et%20al.md#dependency-resolution), then running Build.pl and following the rest of the normal build procedure (as detailed at the end of Build.pl). I did end up needing some additional libraries through apt as well.

guysoft, 2020-06-15:
@RyanMorash Can you list the commands you ran here by any chance to simplify the process of building a script. Moreover those instructions include the GUI side, we need a smaller part of that.

RyanMorash, 2020-06-16:
I’ll need to take a closer look as I can’t remember the exact steps I took. It looks like we could probably use these AppImage build scripts as a reference: https://github.com/davidk/PrusaSlicer-ARM.AppImage

guysoft, 2022-12-04:
FYI
https://github.com/OctoPrint/plugins.octoprint.org/issues/1058