guysoft/OctoPi #446: OctoPi will not pull Slic3r Prusa Edition gcode from the watched folder but oddly will pull Cura gcode from the watched folder.

SeanElliott7, 2017-12-03:
I hope someone can look into this and see what the problem is, I would use the slic3r's 'send to printer' option but I'm running octopi through a port that isn't 80 so it's a nogo at the moment.

Slic3r details:
Slic3r Prusa Edition
Version: 1.37.2-prusa3d-win64
Build: Slic3r-1.37.2-prusa3d-win64-full-201710211408
Operating System: MSWin32
System Architecture: MSWin32-x64-multi-thread
Windows Version: Microsoft Windows [Version 10.0.15063] 

Octopi details:
OctoPrint: 1.3.5 
Raspberry Pi 3

Cura details:
Cura 2.6.2

ntoff, 2017-12-03:
I'm pretty sure Octo*Print* is responsible for pulling in gcode files out of the watch folder.

>I'm running octopi through a port that isn't 80 so it's a nogo at the moment.

Can't you just set the port with hostname.suffix:port in slic3r's settings?

Either way if octoprint isn't pulling files from the watch folder, you might want to report that on the octoprint tracker https://github.com/foosel/OctoPrint/issues/ and fill out the issue ticket in its entirety with all the logs it requests of you. It would also be nice if you provide more info about your setup like is the watch folder on a USB thumb drive, if so, is it ever removed, is it on a network share, how do you get your files into the watched folder, and if you could also supply a sample gcode file that doesn't work, to assist in testing. Thankyou.

Also check your file suffix, are you saving as .gco? .gcode? .gc? etc. and does cura save as the same file suffix?

guysoft, 2017-12-04:
This is an OctoPrint issue, not an OctoPi, issue, please post it to https://github.com/foosel/OctoPrint/issues/

Or report it to the appropriate plugin

closed 2017-12-04 by guysoft