guysoft/OctoPi #447: How to delay OctoPrint startup?

ismangil, 2017-12-09:
I'm tying in my Raspberry Pi running OctoPi to my printer power supply. So I want to delay the startup of OctoPrint so the printer board is ready.

How do I do that?

John-Mc, 2017-12-11:
I don't know if this will point you in the right direction, but when I was using a script to automatically email me the Pi's IP address upon boot up, I needed to delay the running of the script to allow the Pi time to connect to the WiFi network prior to making the attempt. I ended up adding a `sleep(15)` command to the /etc/rc.local  file. I added it right after the initial comment lines (the lines beginning with the #). This made the Pi pause for 15 seconds (you can put in whatever number you want) before proceeding with its startup tasks.

This was a while ago, and the email script author eventually added it to his script, so I've removed it from my rc.local file, but it worked for me.

I'm not sure how well versed you are in making these sorts of changes. I'm far from an expert myself, but if what I wrote does not make sense to you, I'll try to explain further. Let me know if it works out for you.

ismangil, 2017-12-11:
Sounds reasonable. I can look into rc.local.

Thank you.

guysoft, 2017-12-16:
Well, with systemd you can do something like ``After=someserivce``
But at the moment OctoPrint starts with init.d so there isn't such a tool do do this.

How do you know the printer is up?

See:
https://github.com/guysoft/OctoPi/blob/devel/src/modules/octopi/filesystem/root/etc/init.d/octoprint

guysoft, 2017-12-26:
Closing if no reply

ismangil, 2017-12-28:
@guysoft based on discussion on https://github.com/foosel/OctoPrint/issues/2311 I'm trying to wait until the printer port /dev/ttyUSB0 is available

guysoft, 2018-01-04:
Ok, so still should I close if its a different issue?

ismangil, 2018-01-04:
Yes, please close this, there is now no need to delay OctoPrint startup.

closed 2018-01-04 by guysoft