guysoft/OctoPi #448: Problem with OctoPi image download ?

John-Mc, 2017-12-10:
I'm having trouble with the download of OctoPi 0.14 downloaded from https://octoprint.org/download/

It seems to download normally, and double clicking on the zip file expands it as expected. When I try to mount the image, I get the error message: "2017-06-21-octopi-jes image not recognized". (running a Mac, OS X 10.12.6)

I've downloaded 0.13 a number of times as well as a couple of versions of the 0.14 release candidates with no problems. I'm not sure what is going on here. Have others reported problems?

John-Mc, 2017-12-10:
Update: I downloaded a copy of the 0.13 image, and it works fine.

guysoft, 2017-12-26:
Did you try the nightly? The nightly might also solve it.

John-Mc, 2017-12-26:
Yes, the Dec 23, 2017 nightly mounts just fine when I double click on the freshly expanded version, as did all the 0.14 release candidates I downloaded. Seems as though the only one that doesn't is the 0.14 stable release.

guysoft, 2017-12-26:
Sounds like a cashing issue

closed 2017-12-26 by guysoft

John-Mc, 2017-12-26:
Not sure about the caching issue. Multiple downloads had the same problem. Deleted the old file, didn't delete the old file and downloaded a new one, tried on a different computer (also a Mac). All had the problem.