guysoft/OctoPi #449: Add Havit Webcam to additional_brokenfps_usb_devices in octopi.txt

B1717, 2017-12-10:
As per request (~line 39) in /boot/octopi.txt, the following webcam needs special attention.
Manufacturer: Havit
Model Number: HV-N5086

This webcam, originally shows no video streaming output.  Adding its Device ID - taken from lsusb - to octopi.txt results in a video stream from the webcam.  I.e.
In octopi.txt, around line 42, change to 
additional_brokenfps_usb_devices=("046d:082b" "1908:2310")

Thanks

guysoft, 2017-12-11:
Is this the ID of your camera: ``1908:2310`` ?

B1717, 2017-12-12:
Hi,
The id is taken from lsusb.  A full list is show below.

pi@octopi:~ $ lsusb
Bus 001 Device 005: ID 1908:2310 GEMBIRD
Bus 001 Device 004: ID 7392:7811 Edimax Technology Co., Ltd EW-7811Un 802.11n Wireless Adapter [Realtek RTL8188CUS]
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp.
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

Let me know if you need different info.
Cheers,

closed 2017-12-26 by guysoft

firla, 2020-03-12:
Please add also the [Genius WideCam F100](http://us.geniusnet.com/product/widecam-f100):

```
$ lsusb | grep "KYE Systems"  # Genius WideCam F100
Bus 001 Device 004: ID 0458:708c KYE Systems Corp. (Mouse Systems) 

$ fgrep additional_brokenfps_usb_devices /boot/octopi.txt
additional_brokenfps_usb_devices=("046d:082b" "1908:2310" "0458:708c")
```

guysoft, 2020-03-13:
@firla Could you pull request this or open a separate issue so I can reference it in the commit?

F-Light, 2021-09-28:
Thank you for sharing, which helped me solve the problem.

commit 83d0e0da (2017-12-26): Fixes #449