guysoft/OctoPi #451: No wlan0 on new install

orion411, 2017-12-17:
No wlan0 on new install

How do I tell if my WiFi chip is bad on a new board? I've tried for 7 hours to get WiFi working, but no luck. Even tried latest Rasbian Stretch and it doesn't work.

My SSID and password are configured in  /boot/octopi-network.txt and I also tried /boot/octopi-wpa-supplicant.txt.... I tried using the login for my 2G router and my LG G6 phone as a hotspot, but there's no wlan0 interface showing up. Also changed from WPA2 to WPA.  Tried to configure it directly in /etc/network/interfaces. 

Am I doing something wrong or should I try a new board?

guysoft, 2017-12-26:
What is the output of dmesg? It might be a firmware issue.

guysoft, 2018-01-02:
Closing if no reply

closed 2018-01-29 by guysoft