guysoft/OctoPi #452: PL2303 @ 25000 baudrate - Driver problem

gregopti, 2017-12-19:
I wanted to connect 3d printer that uses PL2303 serial inferface to my Rpi with Raspbian but I notice there is problem. Simptoms seems like baudrate is not correct as output are some strange letters. But when I connect 3d printer to PC with Windows it works perfectly.

After reasearching the web it seems that it is problem with linux PL2303 driver that doesn't support baudrate 250000bps that printer operates on.
See: [https://stackoverflow.com/questions/17000078/prolific-pl2303-serial-port-to-250000bps](url)

There have been some suggestions how to fix the driver, but I would like to know if there is any fix already for Octopi.

Printer that I use is: Anycubic i3 mega

guysoft, 2017-12-19:
We had a fix for pyserial at the time. but I don't know about this driver. I suggest you flash OctoPi and try it out.

gregopti, 2017-12-19:
I'm using Octopi on Rpi and in terminal it is receiveing corrupted receive lines like this:
Recv: ���\x0e�
Recv: ���\x0e�
Recv: ���\x0e�

gregopti, 2017-12-21:
Anycubic offered a custom firmware that uses 115,2kbps serial communication. We tested it and it works... at least it reads lines well and accepts orders send to it. There are some other issues regarding gcode communication protocol. I will open a ticket under OctoPrint page.

closed 2017-12-21 by gregopti

guysoft, 2017-12-21:
If they offer a way to add thia kernel to Raspbian I can add a script that installs it to OctoPi

gregopti, 2017-12-21:
They have just sent me file with firmware - "update.zw" and instructions to put it on SD card, insert it in printer and reboot the printer. After some time new firmare was installed. This firmware sets PL2303 serial interface to 115,2kbps and I'm then able to connect using by setting 115200 baudrate on OcroPrint.

electron23, 2018-01-03:
@gregopti Do you found a solution? Where can I find the firmware and how to install it?

guysoft, 2018-01-03:
Perhaps they can share that printer firmware

electron23, 2018-01-03:
I already stay in contact with anycubic support and hope they will share it.

gregopti, 2018-01-04:
I guess I can share it here, since it is the same firmware, with only difference that it works on 115.2kbps.
Unzip file, Copy the modified firmware (update.zw) to SD card , plug it into the printer and then reboot the printer, wait for the firmware update is complete. Remove file from SD cards afterwards.
[update.zip](https://github.com/guysoft/OctoPi/files/1602733/update.zip)

electron23, 2018-01-04:
Perfect, thanks. Did you have the new Ultrabase Version or the old one?

gregopti, 2018-01-04:
I have Ultrabase Version of printer without auto leveling sensor.

gregopti, 2018-01-04:
**NOTE:** There is ticket opened on OctoPrint page, since the problem is not only in communication speed, but also with communication protocol and how OctoPrint handles it.
Link: https://github.com/foosel/OctoPrint/issues/2309

esantoro11, 2018-01-06:
There are two versions of the i3 Mega Ultrabase out there. Most of them are the Mega03 with an 8-bit Trigorilla controller board with firmware v1.1.0. Rarer is the newer version, though may not be better. This version is the Mega04 with a 32-bit Trigorilla Pro controller board with integrated SD card reader and firmware v1.4.

esantoro11, 2018-01-06:
I can confirm that the Mega04 uses integrated Allegro A4988 stepper drivers and the USB software driver for this board is PL2303. The USB software driver for the Mega03 is CP2102.

danitlse, 2018-01-10:
@esantoro11 Hello, can you confirm that the old version of the i3 mega ultrabase is the mega03? Because I bought one on Aliexpress (which does not carry the reference mega03) and I have the last motherboard trigorilla pro, and I wanted to change the steppers to put tmc 2208, I went back to Aliexpress and I saw that there was an announcement with mega03, I would like to reorder order but do you think I would have the old motherboard? thank you very much, and excuse my English I am French :)

esantoro11, 2018-01-10:
@danitlse   Don't trust anything on Aliexpress that states Mega03. You could be sent either the Mega03 or the Mega04. If you look at the box your Trigorilla Pro i3 Mega came in, there is a sticker on it that states Mega04. Contact the Aliexpress seller and state that you need the Mega03 with the v 1.1 firmware, not the Mega04 with the Trigorilla Pro board and the v1.4 firmware. Tell the seller you need a box with a label that states Mega03.

Commander112, 2018-01-13:
Now im a little bit confused...
Is the Firmware-Update @gregopti posted a few days above for the 1.1 or the 1.4 Firmware??

I´ve two Anycubic i3 Mega one with 1.1.0 worked perfect with OctoPrint and one with 1.4.0 that wouldn't work with OctoPrint. 1.1.0 uses CP2102 and 1.4.0 uses PL2303 
I won't flash the wrong Firmware to my 1.4.0 and destroy them ;-)

esantoro11, 2018-01-14:
Commander112,

I believe that the firmware update that @gregopti  posted is for the 32-bit Trigorilla Pro board with 1.4 firmware. I also have two i3 Mega Ultrabase printers. The 1.1 firmware version is known as the Mega03, and the 1.4 firmware version with 32-bit Trigorilla Pro board is known as the Mega04. This is one way that Anycubic 3D distinguishes between the two. This has been confirmed by Alice Yu of Anycubic 3D.

gregopti, 2018-01-22:
For those who were asking about printer version, here is my answer:
The original firmware that I had was V1.4.1RC1/P5 which is then after uploading patched firmware addressed as V1.4.1RC2/P5

I cannot guarantee that this version will work with other firmware/software combination.