guysoft/OctoPi #455: incorrect IP address in octopi-network.txt

HassanKanj, 2018-01-02:
Hello,

in octopi-network.txt (L.54), it should be 192.168.250.1 instead of 192.168.250.10 to match with L.60

guysoft, 2018-01-02:
1. You are welcome to pull request with an explanation
2. What is L.54 and L.60 ? I can't find anything when I Google for it. .1 tends to be the gateway and not the IP of OctoPi

John-Mc, 2018-01-02:
“line 54” and “line 60”?  I don’t have the files handy to see if that makes
sense.

guysoft, 2018-01-02:
Filles are [here in the network module](https://github.com/guysoft/CustomPiOS/blob/CustomPiOS/src/modules/network/filesystem/boot/custompios-network.txt)

guysoft, 2018-01-02:
Line 54: https://github.com/guysoft/CustomPiOS/blob/CustomPiOS/src/modules/network/filesystem/boot/custompios-network.txt#L54

nallar, 2018-01-04:
They're supposed to be different. Line 54 is the configuration to use on a PC. Line 60 is for octoprint.

If you set two systems with the same static IP they won't be able to talk to each other.

guysoft, 2018-01-04:
Yep, ok, closing

closed 2018-01-04 by guysoft