guysoft/OctoPi #456: Raspberry PI3 and 7" LCD Touch screen not powering up

Joaeil, 2018-01-03:
I have a setup involving a CraftBeer PI card mounted on a Raspberry PI3 card and a Raspberry 7" LCD Touch screen.
I´m trying to use 2 power sources, because one is not sufficient to run the whole setup.

The Craftbeer PI card requires 12V/5A and it has an adjustable stepdown "buck" that is set to deliver 5.1V to the Raspberry PI below. The CbPi card/RbPI is powered via a MDR-60-12 transformator (see pictures).
The screen is attempted powered via another MDR giving 5V/3A.
However, the system/screen is not powering up/booting.
Does anyone have an idea of what I´m doing wrong here?

PS! I have measured Volts & Amps using a multimeter.

[image]
[image]
[image]
[image]

Regards
Joachim Eilers

John-Mc, 2018-01-03:
I will defer to Guy here, but since this is not related to OctoPi, I might suggest posting in one of the Raspberry Pi forums such as www.raspberrypi.org/forums/

guysoft, 2018-01-03:
Or CraftBeerPiOS issue tracker: https://github.com/guysoft/CraftBeerPiOS

OctoPi is a distribution running OctoPrint to control 3D printers which I am also developing. Not connected to Craftbeerpi

closed 2018-01-03 by guysoft

Joaeil, 2018-01-03:
Thank you for input @John-Mc and @guysoft !