guysoft/OctoPi #457: Extruder drops from 210 to 195 when printing (PLA)

Martyfree123, 2018-01-04:
Ive noticed that when i head my TAZ 5 to 210 and then start printing, Octoprint automatically drops the temp to 195...

John-Mc, 2018-01-04:
Are you slicing on the Pi, or on another computer? Wherever you are slicing, I'd look at the slicing profile you are using. If it is set for 195, it may override whatever you set via OctoPi's temperature tab

Martyfree123, 2018-01-04:
I'm slicing on my MBP with Cura. I did recently change to a different version of Cura so I'm sure that's the culprit. Thank you so much! I wouldn't have thought of that!

closed 2018-01-04 by Martyfree123

John-Mc, 2018-01-04:
BTW, @Martyfree123, there is an OctoPrint users group on FaceBook where you can also go for help: [OctoPrint Users Group](https://www.facebook.com/groups/piprint/)

Martyfree123, 2018-01-04:
Awesome! :)