guysoft/OctoPi #458: wifi drops

jon012198, 2018-01-06:
so i am using octopi 14.0 and octoprint 1.3.6 and it seams like it is not writing  the network to the network file in linux i try to re start the pi dose not work the wifi work then it just drops and wont reconnect dont know if this is octopi problem or octoprint it worked before i update the pi to 1.3.6octoprint

guysoft, 2018-01-07:
Was a little hard to read because there is no punctuation in the report.
1. Are you using a 2A power supply with a GOOD usb cable?
2. What version of Raspebrrpi?
3. Are you using a wifi dongle or the built-in?

jon012198, 2018-01-07:
so i am using a pi 3 with the built in wifi  and a very good cable with a 4amp charge

guysoft, 2018-01-07:
Does normal Raspbian work?
Sounds like an OctoPi problem,  but I can seem to find what is special.

@foosel?

jon012198, 2018-01-07:
yes raspbian will work if i install desktop  it started happen ever since i started using octprint 1.3.6 old version 1.3.4 and 1.3.5 worked fine when i was using octpi 13.0 . i think it might be since she added the reconnect it sound to me like it is not saveing the network ssid and pass word right

jon012198, 2018-01-08:
i finally got it to get connected to my network anything you want me to look at

guysoft, 2018-01-08:
I need to know what changed before and after and what caused it to connect

jon012198, 2018-01-09:
So all I did was let it sit for a day with no power then plugged it in to power and printer and the cam is a pi cam. I did updated the camera option in octoprint.txt file this line camera_raspi_options="-x 1280 -y 720 -fps 30 -quality 95"

guysoft, 2018-01-09:
You might be hitting power problems if you are powering both the camera, printer and wifi from the pi. So I'd test that, maybe the camera is down now. Apart from that nothing I can do really.

ha1f3mpty, 2018-01-22:
jon012198,

Thank you for posting this. I thought I was going crazy. I have the same hardware setup you do and I too have WiFi issues.

Are you accessing from a mobile device like an Android phone or tablet? My issue only happens when I access from Android, don't have an Apple device so cannot test.

I can SSH to the pi all day long any fay of the week without issue, but when I access the Octopi web page (any tab) the network stack of the pi gets killed some how. If i access from a laptop or desktop (Windows 10) no issues. I know the pi doesn't crash all the way because I can access from attached keyboardand monitor, I restart the network service and it works again. Any ideas or things to try troubleshooting? Oh when I hard wire to Ethernet no issue. 

After reading the net it does sound like a power issue but like  jon012198 said 4amp power supply is more than what the pi should pull and the pi camera should not be that heavy of a load. I have seen where many have issue with the tinny onboard WiFi for most anything. I will try with an external WiFi with a powered hub to see what that does and report back.

guysoft, 2018-01-22:
Ok, if anyone can reproduce on raspbian we can foward the bug to them an the rapsberrypi foundation team. They might be able toto handle power management on this scale.

ha1f3mpty, 2018-01-22:
With raspbian not having a native web server function I don't know how we can reproduce without the possibility of "contamination" from some other product. What web server does octopi use?

guysoft, 2018-01-22:
Flask, but it does not have a mjpgstreamer instance neither. I guess we can try and report somewhere.

jon012198, 2018-01-27:
so i think i found the problem for it if i put my channel static it  just works no problems but once it not a static channel it bugs out

guysoft, 2018-01-27:
It might be your router is switching to channels not supported by Pi due to regional settings. See #348

jon012198, 2018-01-27:
my router only goes to channel 1-11 so dose my pi

guysoft, 2018-01-27:
No idea, perhaps the pi has some bug when the router changes channels in the middle.  Beyond the scope of OctoPi

jon012198, 2018-01-27:
where would i report that and reference this thread

guysoft, 2018-01-27:
I'm not sure. Perhaps: 
https://github.com/raspberrypi/linux
But its probably really hard to debug and you might want to just use a wifi dongle.
Also wifi dongle will tell you if its a driver problem or OS problem.

guysoft, 2018-01-29:
Closing,  since this clearly outside OctoPi's scope.

closed 2018-01-29 by guysoft

ha1f3mpty, 2018-02-07:
I get that this is closed but want to add my final input. I have been running my setup now with the onboard WiFi disabled and using a Alfa AWUS036H 1000mW 1W 802.11b/g USB Wireless WiFi network Adapter with 5dBi Antenna. Just connected right into the pi, no powered sub or anything. It's rock solid. So the onboard WiFi is just crap is all. Probably ok for short range web surfing and stuff but external for any real work.

lenne0815, 2018-12-03:
Just to add in, it has been indeed the auto channel mode for 2.4ghz on my router, once disabled and set to a known good band the raspberrys wifi driver no longer crashes.

graneyc, 2019-01-11:
Another vote for auto channel being the issue. This seems like a reproducible "feature request" that should be logged somewhere to me...