guysoft/OctoPi #460: Build fails: could not open moddep file '/lib/modules/4.4.38-v7+/modules.dep.bin'

kjcole, 2018-01-12:
I've just upgraded Raspian to the latest, and started following the instructions here for building OctoPi.

```
...
Saving to: ‘2017-11-29-raspbian-stretch-lite.zip’

2017-11-29-raspbian 100%[=====================>] 347.83M   951KB/s   in 5m 57s 

2018-01-11 21:44:32 (999 KB/s) - ‘2017-11-29-raspbian-stretch-lite.zip’ saved [364724005/364724005]

pi@flir:~/github/OctoPi/src/image $ cd ..
pi@flir:~/github/OctoPi/src $ ../../CustomPiOS/src/update-custompios-paths
pi@flir:~/github/OctoPi/src $ sudo modprobe loop
modprobe: ERROR: ../libkmod/libkmod.c:557 kmod_search_moddep() could not open moddep file '/lib/modules/4.4.38-v7+/modules.dep.bin'
```

guysoft, 2018-01-12:
Building on the Pi is like building pn a custom environment in this sense, I have no way to know why your kernel is behaving like that.
I really suggest [using the vagrant build method mentioned in the README](https://github.com/guysoft/OctoPi#building-using-vagrant)

kjcole, 2018-01-12:
So, it's a documentation bug.

kjcole, 2018-01-21:
In my case there were two issues:

1. The "latest" raspbian image was "too late".  It downloaded 2017-11-29-raspbian-stretch-lite.zip but

```
$ lsb_release --all
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 8.0 (jessie)
Release:	8.0
Codename:	jessie
```

Using 2017-07-05-raspbian-jessie-lite.zip (the latest "jessie") got further but revealed

2. It needs python3 which was not installed. After adding python3 it appears to get through the procedure.

guysoft, 2018-01-21:
So you were missing python3? Thats listed in the README of CustomPiOS as a requirement

kjcole, 2018-01-21:
Ah. Nothing in https://github.com/guysoft/OctoPi/blob/devel/README.rst indicated a need to read the documentation for CustomPiOS.  The instructions merely tell me to git clone CustomPIOS.  I did:

```
sudo apt-get install gawk util-linux realpath qemu-user-static git p7zip-full

git clone https://github.com/guysoft/CustomPiOS.git
git clone https://github.com/guysoft/OctoPi.git
cd OctoPi/src/image
wget -c --trust-server-names 'https://downloads.raspberrypi.org/raspbian_lite_latest'
cd ..
../../CustomPiOS/src/update-custompios-paths
sudo modprobe loop
sudo bash -x ./build_dist
```

guysoft, 2018-01-24:
Can be updated, I highly recommend using the Building Using Vagrant in the readme. It tends to solve problems like that.

commit bc762d75 (2018-01-24): Fixes #460 documintation bug
closed 2018-01-24 by guysoft