guysoft/OctoPi #462: cannot seem to connect over the internet, no matter what I do

AmorOdium, 2018-01-17:
Hello, I have a new 3d printer I am attempting to set up. it's connected to octopi fine, and the local web interface works fine and is responsive. But I cannot seem to get it working over the internet/remotely. 

Settings/applications involved I know will be relevant:

1. using the noip DUC to provide DNS , this part, at least is working. the noip site resolves to my home router on both :80 and :443 with port forwarding off. but with port forwarding on I get a cannot resolve error.

2. haproxy comes by default, I have tried the default configs that come with the newest octopi install, the suggested ones in the documentation on the octoprint site, and one I found on another forum suggested for a similar problem. with all of these local access through the IP has been fine, but none have let me connect over the internet. 

please let me know what I'm doing wrong here, or if there is more information I need to provide to help you help me.

guysoft, 2018-01-17:
I really can't solve this for you because its a router forwarding issue and is not part of the software.
I can only give tips:

I warmly recommended the Telegram plugin to access remotely. It also has a decent security system in place. That's what I use.

Try connecting from an external network eg your smartphone. Forwarding might not work within your lan

hazrpg, 2018-01-18:
What settings have you got in your router's port forwarding? You might have a conflict somewhere. You didn't also tell us what router you have.

S1CAR1US, 2018-01-20:
there was a time when people posted crap like this on GitHub as an issue, they would be told to go F... well told to go (figure it out) themselves... I feel so bad for these developers, I scrolled through about 60 or 70 of the latest "issues", and it's all just idiots asking for help because they are too stupid to use Google.  at best these are support requests, not issues with development or bugs...   figure it out you dumbfks

guysoft, 2018-01-20:
@S1CAR1US Its true that people don't read. However the reason that we are seeing an increase of non-bug questions is because the size of the community is growing, there are hundred of thousands of downloads to OctoPi at the moment, and some new users just have no place to ask. It means the entry level to 3D printing is also becoming lower, once you had to know a lot in order to start.

Its true i'd love to direct users to a place for basic questions, and I am constantly tempted to post back "Let me google that for you" links. But the truth is people posting here basic questions just means they don't have anywhere else. Or just forms became unpopular, we need to use those more.

guysoft, 2018-02-07:
Closing if no reply

closed 2018-02-11 by guysoft