guysoft/OctoPi #464: no wireless internet connection

trajesus, 2018-01-22:
I just got a raspberry pi zero w to run octoprint in my prusa i3 mk3 and after inserting the sd card with octoprint, the pi cant connect to the internet.
Running the latest version of octopi (0.14) what do I do?

John-Mc, 2018-01-23:
Did you watch the set-up video? There is on on the OctoPrint website, as well as a bunch of them on youtube. 

Are you trying to connect to the internet via WiFi, or over an ethernet connection?

If WiFi: Did you edit the octopi-network.txt file (if you are primarily logging on to just one network) or the octopi-wpa-supplicant.txt file (of you need to long on to different networks)

trajesus, 2018-01-23:
Did not see the video this time because this is the third time setting octopi up, the only difference is that last two times were on a raspberry 2 with wifi adapter and this time is the zero w.

I am trying to connect via WiFi, configured the octopi-network.txt file as I did for the pi 2 and after inserting to de zero w it didn't connect. 

At that moment I thought it could be a problem with the setup I did, to confirm if everything was ok I inserted the sd in a pi 2 and it worked.

trajesus, 2018-01-23:
I tried a wifi usb dongle and it works that way, but is it a problem on the octopi or my particular raspberry?

John-Mc, 2018-01-23:
@trajesus the internal WiFi antenna is much weaker than most of the USB WiFi dongles. As an experiement, you can move your Pi0W close to your WiFi router/access point and see if it picks up. 

You could also try entering the command `sudo   iwlist wlan0 scan`  This displays all of the Wifi access points that your Pi can see, even if it is not connected to them.

I don't want to step on Guysoft's toes, since this is his page, but if your card works in another Pi, that demonstrates that the problem is not with OctoPi. This site is mainly for reporting addressing bugs or suggesting added features to OctoPi. Since you don't appear to have an OctoPi bug, you may want to try one of the sites dedicated to trouble shooting set-up problems with OctoPi/OctoPrint or the PI in general. Here is a fairly active facebook page where a lot of folks have found help getting octoprint up and running on their Pi's: [https://www.facebook.com/groups/piprint/](url)

trajesus, 2018-01-23:
I did what you said and moved it closer, didn't know the antenna was much weaker, now it worked :D 
Also, I want to apologize, I didn't know there was a group for troubleshooting set-up problems with OctoPi :/ thank you for the help :D

closed 2018-01-23 by trajesus

John-Mc, 2018-01-24:
Hey, no worries. I'm not the arbiter of this site anyway. I just thought that other one might be of interest to you. There are a LOT of people on there, and you'll get multiple answers to questions, sometimes within minutes.

that sudo iwlist scan...  command can help troubleshoot. In addition to listing WiFi networks the Pi can see, the printout includes info on signal quality and strength.

guysoft, 2018-01-24:
@John-Mc No but you are a great help :)
Good FOSS projects have the guy who replies before the devs do. 
Second what @John-Mc said so far

John-Mc, 2018-01-24:
Hey, I can't do the programming, but maybe I can help in other ways.

trajesus, 2018-01-28:
By the way, just a question @John-Mc and @guysoft , how is the octopi-wpa-supplicant.txt file configured? :o always used the normal octopi-network.txt file but could be handy to configure several networks :)

John-Mc, 2018-01-28:
It's pretty simple. Just comment out the lines you modified in octopi-network.txt (since you will no longer be using this to connect) and save that file. Then open and edit the octopi-wpa-supplicant.txt file, uncomment **4** lines and fill in your SSID and password. If you want to add another network, simply duplicate that block of lines and fill in the SSID and password of that network. For future reference, I usually add a comment line above each block with some information on that network (e.g. '# McNerney Home Network' or '# Monkton Central School Network').  

OctoPi will connect to whatever network it sees when it boots up. It's **really** handy if you move your printer from place to place (I volunteered in a school, so there was a lot of back and forth).

I haven't tried it yet in an area where it can see more than one known network: does it just pick the first one on the list? the strongest signal? I haven't run into that yet. It would be cool if the list order dictated the connection preference, but maybe it makes more sense to pick the best signal (at the school, the strongest signal was also the busiest network, so I didn't bother putting that one in the octopi-wpa-supplicant.txt). I don't think it can give you a dialog box to choose, since if you are operating headless (as many OctoPi are), you don't have access until **after** it connects.

trajesus, 2018-01-28:
thank you :D I will try to configure that way :)