guysoft/OctoPi #465: Cannot figure out how to add my cura profile to octoprint

Carter011, 2018-01-27:
First off, this may be an insanely stupid question but I am completely new to all things 3d printing. I don't know how to save my profile a .ini file. I'm using the newest version of cura and see no option for it. Any help would be appreciated.

John-Mc, 2018-01-27:
If you want to use it for slicing in OctoPrint, you have to create the profile in an older version of Cura: use version 15.04 (or older). If you log into OctoPrint as an admin user, and click on the little wrench icon at the top to go into settings, and scroll down the sidebar until you see the entry for CuraEngine, you'll see a notation after the name of <=15.04. If you click on that, you'll get a bit of an explanation.  

If you want to use the newer versions of Cura, you'll have to run them on your Laptop/desktop computer. Fortunately, Cura 3.x has the ability to tie in directly with OctoPrint, so you slice on your computer, and send the gcode to OCtoprint and control your printer all from within Cura

If you want to slice using OctoPrint on your Pi, You'll need the older version of Cura found at [https://ultimaker.com/en/products/cura-software/list](url)   And yes, version 15.x is older than version 2.x or 3.x. - Ultimaker changed their numbering scheme when 2.0 came out.

I noticed that the note when you click on CuraEngine in OctoPrint says 15.06 and newer is NOT compatible, but there is no 15.06 listed in the Utimaker archive. There are versions 15.04 through 15.04.6 listed for various operating systems. I **believe** 15.04.6 will work (and you don't have to drop all the way back to 15.04), but I am not certain. Perhaps @guysoft or someone else can confirm?

Once you have loaded the correct version on your desktop, launch it, and put in your desired profile settings. You may have to enable "expert" or "full settings" to reach all the variables you might want to tweak. I believe you also need to do "add new machine" and adjust machine settings for your printer as well? When everything is done, choose File>Save Profile... and export your profile as a .ini file. [Note: I'm using  customized version of Cura for Lulzbot printers. I'm not sure if the menu commands may have been customized as well, but hopefully they are close enough that you can figure it out.]

The downside of you OctoPrint implements it's Cura plugin is that you need to import a new profile for every little tweak you want to make. If you would like to be able adjust some of these settings from within OctoPrint, try adding the "Full Featured Slicer" plugin. This adds a tab to OctoPrint which allows you to tweak many of the commonly used filament profile settings from within that tab.

guysoft, 2018-01-29:
Indeed, use version 15.04.
@foosel you think its time to upgrade this somehow?

John-Mc, 2018-01-29:
I'm hoping someone makes anupdated Cura plugin. I just got word that I have to flash my Lulzbot Mini's firmware to be able to use Cura 2.6.xx or later. Once I flash it, STLs sliced under older versions will not print. I'm assuming this means I can no longer use it with the version on OctoPrint. This will seriously disrupt the workflow at the elementary school where I volunteer - none of their ancient laptops will run much of anything in terms of demanding software. OctoPi was a lifesaver for us in the classroom.

And sorry, @guysoft . I know this is topic drift... not really an OctoPi issue.

foosel, 2018-01-29:
To quote myself from [around a year ago](http://www.reddit.com/r/3Dprinting/comments/5f8jdy/-/daiq0a7):

> Cura 2 hasn't been around that long. Attempting to integrate with it before (and when I started trying that i even still had help from my then-employer) started about 1.5 years ago and ended basically with having to scrap everything every other month before the (undocumented) engine parameters and profile structure was changed ever so slightly again. The hierarchical profiles also make an integration/reuse of those extra tricky (you need all files a profile depends on, not just a single file - if they finally introduced a batteries-included format that I don't know of yet though, that would be good news).
> 
> In a nutshell, about a year ago I decided to scrap efforts to have support for the new cura engine (now known as cura 2, back then targeted to be cura 15.10) in 1.3.0, simply because it cost too much lost time trying to hit a moving target.
> 
> Now that the waters appear to have become a bit more smooth that could be reinvestigated, but by the time that had happened OctoPrint was already in feature freeze mode for 1.3.0
> 
> Slicer support is also very rudimentary (for the reasons for that see OctoPrint on Air episode 4 on YouTube) and not the key focus of the software. Which means it will probably stay low priority (I already got complains for even adding it). With that being said, slicer support in OctoPrint is fully based on plugins. If people are actually sorely missing built in support for the new cura engine and don't want to wait for me to add support for that (hint: no timeline yet), the documentation on SlicerPlugins and looking at the source of the existing cura plugin (which I btw never should have bundled in hindsight) should really provide enough information for writing a cura2 compatible plugin and being the hero of everyone waiting for this :) At this time, I'm not aware of anyone doing this actively though - a quick message on the mailing list or as a brainstorming ticket on the issue tracker would be nice for coordinating if anyone tackled this.
> 
> tldr: first attempts to integrate boiled down to shooting at moving targets, and when stuff finally became stable other things had way way higher priority and this will probably not change in the near future. Everything slicing related is plugin based anyhow, if you need it now, write your own, docs and examples are there.

To add to this, I'm thinking about unbundling the plugin (with a wizard that triggers if existing slicing profiles are detected, offering to immediately install it from the repository). Since now both Slic3r *and* Cura offer automatic upload to OctoPrint, I would imagine that even less people slice directly on the server (and let's face it, no matter how much time and effort I put into slicing in OctoPrint, it will always be behind in terms of user experience compared to a dedicated program). I'd leave the 15.04 engine in OctoPi since that makes things easier for people who actually do use it (I know of a couple of classroom settings where it's utilized heavily). Should anyone so up to do the work needed to get Cura Engine 2/3/X integrated, that could also be built in, but so far nobody has to my knowledge.

Long story short: updating to the modern Cura engine is way too much effort to justify for the foreseeable future considering the utilisation and thus low priority of onboard slicing - but that shouldn't keep others from building and maintaining (!) a corresponding plugin, and I'll also happily help there where I can.

John-Mc, 2018-01-29:
Gina, I fully understand. I had read your previous comments and understand your need to remain focused on OctoPrint. I do remain hopeful that someone will start work on a plugin, since doing so if FAR beyond what I am capable of (my last experience with any kind of programming was back in the 1970s and early '80s teaching myself BASIC, and later taking an evening class in Pascal). Due to the firmware issues I mentioned, I'm faced now with staying stuck on Cura 15.04 for my own use so I can also use the printer in the classroom with OctoPi/OctoPrint, or upgrading Cura at home, and abandoning the classroom volunteering (Or I suppose I could reflash the firmware every time I moved between home and the school).

Unbundling the plugin is fine with me, as long as it remains available. I just hope someone does take on the project of making a new one.

foosel, 2018-01-29:
@John-Mc do you have any details on those firmware issues? Because that sounds a bit odd considering that slicers usually do not produce a lot more than lots of G1 and maybe some temperature and homing commands. Feel free to shoot me a private mail at gina -at- octoprint.org on this, I don't want to hijack this ticket further :)

John-Mc, 2018-01-29:
msg sent to Gina. For others interested, scroll down to Download, Installation, and Removal Instructions on this page: [https://www.lulzbot.com/cura](url)

chiques, 2018-12-31:
This is a major PITA. I don't even use Cura, I use Matter Control when directly connected. 
The solution that worked for me was to go to the Cura site, download the 15.04 version, export the profile and then import into Octoprint.

jneilliii, 2020-05-12:
I've been looking into getting the latest CuraEngine into OctoPrint/OctoPi and have succesfully compiled version 4.6.1. This was within Python 3, going to test Python 2 in the near future. If successful I can work on updating the OctoPrint plugin to work with the new profile settings of the CuraEngine so we can get the newer features, etc. Unfortunately, I have yet to see a method to export the Cura settings directly into json format, which is required by the new version, but there may be some tricks on the back-end that maybe @fieldOfView might know more about as he is the author of Cura's OctoPrint Connection plugin and contributes to Cura as well.

guysoft, 2020-05-12:
Ok, for reference I made a comment here, with where in the code the build should go: https://github.com/OctoPrint/OctoPrint/issues/1284#issuecomment-627148312

fieldOfView, 2020-05-12:
> I have yet to see a method to export the Cura settings directly into json format

There is none. I think you'll need a global json, and a json for each extruder. I can probably create a plugin to create these.

jneilliii, 2020-05-12:
Haven't got that deep into it yet. I just know that when I started trying I kept having to go back to the cura repo and grabbing additional json files until it worked, so I assume there is some forum of dependency linking within them?