guysoft/OctoPi #466: 17.04 reached End of Life please change

Lino77, 2018-01-28:
lino@asrock:~/OctoPi/src/vagrant$ sudo vagrant up
Bringing machine 'default' up with 'virtualbox' provider...
==> default: Importing base box 'ubuntu/zesty64'...
==> default: Matching MAC address for NAT networking...
==> default: Checking if box 'ubuntu/zesty64' is up to date...
==> default: Setting the name of the VM: vagrant_default_1517154167958_89369
==> default: Clearing any previously set network interfaces...
==> default: Preparing network interfaces based on configuration...
    default: Adapter 1: nat
    default: Adapter 2: hostonly
==> default: Forwarding ports...
    default: 22 (guest) => 2222 (host) (adapter 1)
==> default: Running 'pre-boot' VM customizations...
==> default: Booting VM...
==> default: Waiting for machine to boot. This may take a few minutes...
    default: SSH address: 127.0.0.1:2222
    default: SSH username: ubuntu
    default: SSH auth method: password
    default: Warning: Remote connection disconnect. Retrying...
    default: Warning: Connection reset. Retrying...
    default: Warning: Remote connection disconnect. Retrying...
    default: Warning: Authentication failure. Retrying...
    default: Warning: Remote connection disconnect. Retrying...
    default: 
    default: Inserting generated public key within guest...
    default: Removing insecure key from the guest if it's present...
    default: Key inserted! Disconnecting and reconnecting using new SSH key...
==> default: Machine booted and ready!
==> default: Checking for guest additions in VM...
    default: The guest additions on this VM do not match the installed version of
    default: VirtualBox! In most cases this is fine, but in rare cases it can
    default: prevent things such as shared folders from working properly. If you see
    default: shared folder errors, please make sure the guest additions within the
    default: virtual machine match the version of VirtualBox you have installed on
    default: your host and reload your VM.
    default: 
    default: Guest Additions Version: 5.1.22
    default: VirtualBox Version: 5.0
==> default: Configuring and enabling network interfaces...
==> default: Installing NFS client...
The following SSH command responded with a non-zero exit status.
Vagrant assumes that this means the command failed!

apt-get -yqq update
apt-get -yqq install nfs-common portmap
exit $?

Stdout from the command:

Stderr from the command:

E: The repository 'http://security.ubuntu.com/ubuntu zesty-security Release' does no longer have a Release file.
E: The repository 'http://archive.ubuntu.com/ubuntu zesty Release' does no longer have a Release file.
E: The repository 'http://archive.ubuntu.com/ubuntu zesty-updates Release' does no longer have a Release file.
E: The repository 'http://archive.ubuntu.com/ubuntu zesty-backports Release' does no longer have a Release file.
E: Failed to fetch http://security.ubuntu.com/ubuntu/pool/main/p/python2.7/libpython2.7-minimal_2.7.13-2ubuntu0.1_amd64.deb  404  Not Found [IP: 91.189.88.161 80]
E: Failed to fetch http://security.ubuntu.com/ubuntu/pool/main/p/python2.7/python2.7-minimal_2.7.13-2ubuntu0.1_amd64.deb  404  Not Found [IP: 91.189.88.161 80]
E: Failed to fetch http://archive.ubuntu.com/ubuntu/pool/main/p/python-defaults/python-minimal_2.7.13-2_amd64.deb  404  Not Found [IP: 91.189.88.161 80]
E: Failed to fetch http://security.ubuntu.com/ubuntu/pool/main/p/python2.7/libpython2.7-stdlib_2.7.13-2ubuntu0.1_amd64.deb  404  Not Found [IP: 91.189.88.161 80]
E: Failed to fetch http://security.ubuntu.com/ubuntu/pool/main/p/python2.7/python2.7_2.7.13-2ubuntu0.1_amd64.deb  404  Not Found [IP: 91.189.88.161 80]
E: Failed to fetch http://archive.ubuntu.com/ubuntu/pool/main/p/python-defaults/libpython-stdlib_2.7.13-2_amd64.deb  404  Not Found [IP: 91.189.88.161 80]
E: Failed to fetch http://archive.ubuntu.com/ubuntu/pool/main/p/python-defaults/python_2.7.13-2_amd64.deb  404  Not Found [IP: 91.189.88.161 80]
E: Failed to fetch http://archive.ubuntu.com/ubuntu/pool/main/libt/libtirpc/libtirpc1_0.2.5-1.1_amd64.deb  404  Not Found [IP: 91.189.88.161 80]
E: Failed to fetch http://archive.ubuntu.com/ubuntu/pool/main/r/rpcbind/rpcbind_0.2.3-0.5_amd64.deb  404  Not Found [IP: 91.189.88.161 80]
E: Failed to fetch http://archive.ubuntu.com/ubuntu/pool/main/n/nfs-utils/nfs-common_1.2.8-9.2ubuntu2_amd64.deb  404  Not Found [IP: 91.189.88.161 80]
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?

guysoft, 2018-01-28:
Hum, will change to Artful, but also considering changing to Stretch. 
@foosel  - any preference?

foosel, 2018-01-28:
Completely preference free :)

guysoft, 2018-01-28:
Tried using artful. Got:
```
Failed to start vboxadd-service.service: Unit vboxadd-service.service not found.
Got different reports about installed GuestAdditions version:
Virtualbox on your host claims:   5.1.28                                                                                          
VBoxService inside the vm claims: 5.1.30                                                                                          
Going on, assuming VBoxService is correct...                                                                                      
bash: line 3: setup: command not found
==> default: Checking for guest additions in VM...
The following SSH command responded with a non-zero exit status.
Vagrant assumes that this means the command failed!                                                                               

 setup                                                                                                                            

Stdout from the command:                                                                                                          

Stderr from the command:                                                                                                          

bash: line 3: setup: command not found
```
Stretch does not give that, it gives this error:

```
....] Starting nfs-kernel-server (via systemctl): nfs-kernel-server.serviceJob for nfs-server.service canceled.
 failed!
==> default: Mounting NFS shared folders...
The following SSH command responded with a non-zero exit status.
Vagrant assumes that this means the command failed!

mount -o vers=3,udp 192.168.55.1:/home/guy/workspace/custom_raspberrypi_os/CustomPiOS/src /CustomPiOS
result=$?
if test $result -eq 0; then
if test -x /sbin/initctl && command -v /sbin/init && /sbin/init 2>/dev/null --version | grep upstart; then
/sbin/initctl emit --no-wait vagrant-mounted MOUNTPOINT=/CustomPiOS
fi
else
exit $result
fi

Stdout from the command:

Stderr from the command:

mount.nfs: requested NFS version or transport protocol is not supported
```

Looking for solutions.

guysoft, 2018-01-28:
My bad, had duplicate entries of CustomPiOS mount point in ``/etc/exports``. Since Stretch is working, and Debian replace versions less often, we are moving to stretch. Hopefully it would stay that way because RaspberyPi foundation also use stretch.

commit 303a751c (2018-01-28): Fixes #466 , moved to debian/stretch64 on vagrant, update your VMs
closed 2018-01-28 by guysoft