guysoft/OctoPi #468: i can not connect to octopi in any way, please help

leytanelectronics2, 2018-01-28:
I am using octoprint version 1.3.4 and octopi version 0.14, I can not connect to the raspberry pi in any way and i am utterly lost, nothing I do gets me into the user interface, can;t access it from octopi.local, can't access the ip address, i can't even ping it.

I do not understand what to do and I do not want to go poking around under the hood without knowing what to look for or what to do so any and all help will be grateful.

guysoft, 2018-01-28:
Can you please post:
1. What have you tried - what ways?
2. Assuming you tried wifi can yourl post your wpa-supplicant.txt?
3. Did you try plugging your Pi to a screen and watch that it boots?

leytanelectronics2, 2018-01-28:
to guysoft: i only tried switching between a wired connection, i did plug in a screen to watch it boot, i don;t know what a wpa-supplicant.txt file is or if its on the machine.

guysoft, 2018-01-29:
1. Can you post a picture of the boot?
2. Can you please explain how you set up the wifi?
Please read about octopi-wpa-supplicant.txt file. Its part of the wifi setting instructions

leytanelectronics2, 2018-01-29:
i listed everything, i unchecked the things in the network config and it didn't work, i could ping through wifi but my wifi roughter has app isolation on which means i can;t access any other machine through that roughter, its the lan that's giving me problems.

guysoft, 2018-02-01:
Does your lan connection have dhcp? Does the Pi gets an IP?

guysoft, 2018-02-11:
Closing if no reply

closed 2018-02-21 by guysoft