guysoft/OctoPi #473: Loosing connection/Print stopping mid print.

titchgamer, 2018-02-05:
Sorry not sure if this is the right place to ask for support but I am having a problem with my Octopi set up stopping my printer mid print.

I have had this issue both printing from my PC via octopi and printing from the printers SD via octopi.
The octopi page on my laptop says disconnected then stops the printer and it goes into cool down.

I am using a RPi 3 with the built in wifi connected via the USB lead that came with the printer.

I have attached the log file which I hope may be some help?

Thanks.
[octoprint.log](https://github.com/guysoft/OctoPi/files/1696601/octoprint.log)

MikeBookham, 2018-02-05:
I don't know a lot about Octopi/Octoprint however I'd suggest looking at the line:
2018-02-05 19:44:25,470 - octoprint.util.comm - ERROR - Please see https://bit.ly/octoserial for possible reasons of this.

The URL quoted gives some good pointers to what might be the issue, check each one in turn to see if that helps.

titchgamer, 2018-02-05:
Thanks for that Mike I will try that out!

titchgamer, 2018-02-11:
Well I read through the above link and tried all the steps but ime still having some issues.

I swapped out my PSU for a official Rpi adapter which is rated at 2.5A (couldnt find one rated at 3A).
I also fitted ferrite cores to my USB leads and plugged the web cam into a powered hub, But the printer is still plugged directly into the Pi.

So yeah, It crashed out on me last night but luckily it was only a 1hr print.
The 2 1 hour prints I did immediatly after it failed went fine though so it seems to of reduced but not perfect still.
I am considering using some sugru to effectively glue the USB into the actual printer incase its working loose as I know they printer connections are not meant to be great but other than that ime still none the wiser!

DarthKegRaider, 2018-08-02:
Hi @titchgamer,

A bit late to the reply, but I power my Raspberry Pi via the printer's 12V with a BEC (https://hobbyking.com/en_us/turnigy-5a-8-26v-sbec-for-lipo.html).  It just means that before you physically "Power off" the printer, you need to shutdown your raspberry pi.

guysoft, 2018-08-12:
I am closing this since this is an OctoPrint issue.

closed 2018-08-12 by guysoft