guysoft/OctoPi #474: Octoprint service crashes during print.

speatzle, 2018-02-11:
Last Night i was running my first longer print with Octopi it worked fine the last week with lots of 1h-2h prints. So i decided to step it up with a 6h print (i can't upload gcode files they are to big).
during the print at around 50% the logs and the printer stop. so when i woke up i tried to check my phone. Octopi was not available so i went down stairs where the printer was still sitting at printing temperature and doing nothing.

[image]
[image]

so i went and tried to get into the web interface via my laptop. and i get the octoprint service is not running message. so i ssh into the pi and try the grep python whatever command. nothing. i go to the log nothing https://pastebin.com/pWb9kNkk  (just cuts of like this at the end). so i try htop nothing either.

[image]

and service thinks its "OK"

[image]

but it just isent running.
So i restart octoprint via sudo service octorint restart.
and it comes back. but i have an unusable 50% print.
(there is no issue with my raspberry pi or my printer both are working fine)

####Things i did to my Octopi

1. Plugins i use:
[image]

2. i made symlinks to an external usb drive for the folders logs,timelaps,uploads and watched
3. i ran a simplify 3d gcode file (about 10mb in size).

#### Version of OctoPrint

just installed last week: OctoPrint 1.3.6 running on OctoPi 0.14.0

#### System running OctoPrint

raspberry pi 2

#### Printer model & used firmware incl. version
i am running a ramps 24v modified ctc bizer1 with e3dv6 hotend and marlin version RC-8 CONFIGURATION_H_VERSION 010100

#### Browser and Version of Browser, Operating System running Browser

Firefox/win10

guysoft, 2018-02-11:
Are you using a 5V 2A power supply and a GOOD  power cable?

speatzle, 2018-02-12:
im using the 2.5 amp charger that came with a different raspberry pi 3

the log is very big and gose back to were i first started octoprint so big that i cant upload it to pastbin so i took the last part of the log and uploaded it (also above) https://pastebin.com/pWb9kNkk

guysoft, 2018-02-13:
that output is only from telegram plugin, about z changes. 
There is nothing else there.
You could try disabling that plugin. 
But from this point this is really an OctoPrint issue, not an OctoPi issue.
You should post it in the OctoPrint issue tracker at: https://github.com/foosel/OctoPrint/issues/

guysoft, 2018-06-09:
Any news on this?
Will close if no update ont he telegram issue

closed 2018-06-14 by guysoft