guysoft/OctoPi #476: OctoPrint server is not running according to broswer (firefox).

nissguy, 2018-02-12:
I can't remember what version I had before, but I did have a working version on my pi. Then somehow the microsd got corrupted and I had to start from scratch.

Using Etcher, I put on the 2017-06-21-octopi-jessie-lite-0.14.0.img, and after much screwing around got the pi on the wifi.

It has an IP address and is asking me to connect via my browser, and I get this message:

https://imgur.com/z0uJHiT

When I try the first step to verify the process is running, no joy.
I tried to look for the logs where shown, however there is no logs directory in octoprint, the only directory when is put in the command ls, is OctoPrint, and there is no logs folder in there either.

sudo services octoprint restart does nothing to fix it either.

Any thoughts?

guysoft, 2018-02-12:
Can you:
1. Verify you are using a power supply of 2A and a good power cable?
2. Do what the message asked and provide octoprint.log please

nophead, 2018-02-12:
The log files are in .octoprint, which is a hidden directory because it starts with a dot. You can see it with ls -a

nissguy, 2018-02-12:
@nophead I found the hidden directory for .octoprint, and when I'm in it I used the command ls -a, and ls -all, and the only thing that came up was:
. 
..
config.yaml

No log file.

@guysoft It is in fact a 2A supply, and a good cable. All the same hardware from when it functioned before. As per above, I'm still trying to find said log file.
https://github.com/guysoft/OctoPi/issues/476

guysoft, 2018-02-12:
Is this a new sd card? No log files sounds like sd corruption. You can read from them, but not write

nissguy, 2018-02-12:
Nope, same SD Card as before but as per my OP I wrote the image to it using etcher. I also had to update various config files for the wifi dongle to work. Do SD cards have sectors like HDDs that could be bad?

guysoft, 2018-02-12:
Basically yes. Id try another sd.

nissguy, 2018-02-21:
@guysoft It was indeed the sdcard. Back up and running. Thank you very much for your assistance, now if I could only solve the squeeking.

Cheers!

closed 2018-02-21 by guysoft