guysoft/OctoPi #477: Custom Website in Octopi

noxferatus, 2018-02-13:
Hi, 
I want to create a php website in the octopi raspberry, the thing is that I installed apache2 (I think that's a mistake) but http://localhost always takes me to the octopi screen... is there a way to do this?
by "this" I mean, create a folder, set css, html, php (mysql maybe) files inside and open a website inside the raspberry octopi but keeping the octoprint screen if I go inside "octopi.local" url??

Thanks and sorry for bad english!

guysoft, 2018-02-13:
OctoPi uses a note balancer call HAProxy to display the page. You need to change the HAProxy config in order for your Apache server to work. Or you can or you can remove it altogether and just use OctoPi any online custom port.

noxferatus, 2018-02-13:
Do you know if there's a way, if I change the HAProxy config, to open, for example:
localhost -> octoprint interface
localhost/html -> personal website
and of course, if I connect via "octopi.local" that still takes me to octoprint interface...?

guysoft, 2018-02-14:
Yes, pleaslook at HAProxy config shipped with OctoPi. The answer is there.

guysoft, 2018-02-21:
Closing of no reply

closed 2018-02-28 by guysoft

averter, 2019-12-17:
Hi @FelipeDelgadoR I am also interested in doing this. Can you please tell me if you were successful and have a step by step guide? Thank you very much

guysoft, 2019-12-18:
@averter He seemed to not reply, I'd recommend asking at the OctoPrint community at https://discourse.octoprint.org/

averter, 2019-12-18:
Thanks. This is the closest [thread](https://community.octoprint.org/t/octoprint-website-running-on-apache2/4863/2) I've found. Guess I'll just join the fray