guysoft/OctoPi #478: no longer to get to webcam settings at http://printer:8080/control.htm

ch0de, 2018-02-14:
I reloaded my Octopi few days ago and i am not longer able to get to http://printer:8080/control.htm. 

when i go there it is a page that just give a image and a video steam, the old page that looked like this https://imgur.com/a/zO6Hp     is gone

guysoft, 2018-02-14:
I have no idea. Please provide logs. But it sounds like a localized error.

ch0de, 2018-02-14:
[octoprint.log.2018-02-13.txt](https://github.com/guysoft/OctoPi/files/1725574/octoprint.log.2018-02-13.txt)

ch0de, 2018-02-21:
https://imgur.com/a/zO6Hp

Is that the webpage i should still get when i goto  http://printer:8080/

If so maybe i will reload again.

guysoft, 2018-02-21:
Wait, when you say "reloaded", you mean flashed a new image?
Did you flash a newer version? It might be that the HAPRoxy parameters have changed.

ch0de, 2018-02-21:
Yes I flashed the new image from GitHub.

ch0de, 2018-02-21:
[image]
here is what the screen looks like after i flashed new image..

guysoft, 2018-02-21:
This ks expected behavior due to security reasons as stated inhttps://github.com/guysoft/OctoPi/pull/359

closed 2018-02-21 by guysoft

ch0de, 2018-02-21:
I understand this is now excpected behave after reviewing your other post. However how do i get that back? i only see a discription for the -N option. 

I need that camera page to disable auto focus and set the focus range.

guysoft, 2018-02-24:
The information is here.
I am really not supporting this:
https://github.com/jacksonliam/mjpg-streamer/tree/master/mjpg-streamer-experimental/plugins/input_uvc

To do what you want you need to reverse what Gina discribes. aka switch here the webroot from www-octopi back to www:
https://github.com/guysoft/OctoPi/pull/359/files#diff-90ef84c6aacfc0fe4a6a5655ff158526R22