guysoft/OctoPi #479: UnicodeDecodeError: 'ascii' codec can't decode byte 0x94 in position 48: ordinal not in range(128)

gordonBimig, 2018-02-15:
Hello,

I'm trying to print several items in one print, but the print is failing (simply stops, it seems always on the same layer height). I changed the items arrangement in the slicer and also removed some other parts, but the problem remains. I found the following lines in the log:

2018-02-14 19:56:55,308 - octoprint.util.comm - ERROR - Something crashed inside the serial connection loop, please report this in OctoPrint's bug tracker:
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/util/comm.py", line 1249, in _monitor
    line = self._handle_errors(line)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/util/comm.py", line 2003, in _handle_errors
    level=logging.WARN)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/util/comm.py", line 529, in _to_logfile_with_terminal
    log = message + "\n| " + log
UnicodeDecodeError: 'ascii' codec can't decode byte 0x94 in position 48: ordinal not in range(128)

Hopefully you can explain that to me. Is this a bug?

#### Version of OctoPrint
1.3.6

#### Operating System running OctoPrint
OctoPi version    : 0.14.0

#### Printer model & used firmware incl. version
UM Clone with Marlin1.1.5

#### Browser and Version of Browser, Operating System running Browser
Google Chrome 63.0.3239.132 on Windows 10 x64 1709 16299.125

#### Link to octoprint.log
https://pastebin.com/X0T2kM7Q

### G-Code with which the error occured:
https://www.file-upload.net/download-12979754/spheres_emboitees.7z.html

Hope this is enough information?

guysoft, 2018-02-15:
This is an OctoPrint issue, not an OctoPi, issue, please post it to https://github.com/foosel/OctoPrint/issues/

But its a well written bug report :)

gordonBimig, 2018-02-15:
Ups, when did I get so confused? :-D

Thanks for your reply, will post in the OctoPrint github.

closed 2018-02-16 by guysoft