guysoft/OctoPi #481: Turning off Wifi disables wired?

bcheezum, 2018-02-17:
I'm not sure if this is a bug or something specific to the Raspberry Pi 3 I'm using.
My wifi is 192.168.0.0/24 and my internal wired is 10.0.64.0/24.

I downloaded the latest and used Etcher to write to my SD card. If I edit octopi-network.txt, and add valid wifi (WPA2) and wired static settings, it boots fine. I can ssh to both the wired and wireless IPs.

However, if I disable wifi in anyway (since I want to use wired) it kills all connectivity and I have to revert whatever I've changed.

Ways I've disabled wifi:

ifdown wlan0
commenting out all of the wifi settings in octopi-network.txt
commenting out just the wpa-ssid and wpa-psk lines in octopi-network.txt
added "dtoverlay=pi3-disable-wifi" into /boot/config.txt

Any pointers would be appreciated as I can't seem to figure this out on my own.
I don't *think* this is a Raspberry Pi thing because I can't seem to find any mentions of a similar issue anywhere online.

guysoft, 2018-02-17:
Is this the behavior in Raspbian? Because it sounds like a Raspbian issue.

guysoft, 2018-02-24:
Closing if no reply

bcheezum, 2018-02-24:
Sorry for the delay, I missed the notification for your previous reply. 

I don't believe it's Raspbian behavior because I have another RPI3 running as a DNS server and wifi is disabled. 

However that is on a previous release as far as I know.

guysoft, 2018-02-24:
I can't debug based on your belief,  I need you toto test.

bcheezum, 2018-02-24:
What exactly do you need me to test?

I have tested all of the various ways to disable wifi on an RPI (as mentioned in my original post) and they all result in the same thing when running under octopi, no networking. I have another RPI 3 running Raspbian (Jessie) which has wifi disabled and it's working just fine on wired. However, octopi doesn't behave the same way.

Just as you can't debug based on my "belief" I can't test based on flippant comments with no information. 

If you don't want people coming to you for help with octopi you should probably just disable issues.

guysoft, 2018-04-22:
I need you to test if its happening on raspbian too. 
Just saw now. 
Also as of now you might get a faster reply in the forum at https://discourse.octoprint.org/

supertaz, 2018-05-08:
You're doing it wrong.  If you want to disable wifi, the correct way is to edit /boot/config.txt and add this line:

dtoverlay=pi3-disable-wifi

If you also want to disable bluetooth, add this line as well:

dtoverlay=pi3-disable-bt

If you disable wifi with the overlay, eth0 works just fine, from personal experience.

npbazan, 2018-05-09:
Hi I'm always used octopi with lan and static IP address. Now in this newer version I cannot find to put an static IP.. unfortunately I use new raspberry pi3b+ and they dont allow older versions of octopi. So bad, i cannot solve it..

bcheezum, 2018-05-09:
@supertaz I had tried that previously (see original comment) and it kills all networking.

Just to be sure, I tried it again remotely from work. After a reboot I can't reach the device over a wired connection. I'll have to console it when I get home and see what it's status is.