guysoft/OctoPi #485: WIFI Dongle not Working

Cyphen-Cyn, 2018-02-19:
Okay this is driving me absolutely crazy. i have re-imaged 8+ times and i still cannot get this to work. the wifi dongle works when i image for Astroprint but not for OctoPi. i have followed about 4 different write ups with everyone saying "hey dummy your using smart quotes" not the case here since i am using windows and Notepad++, i have also done it through "sudo nano" keyboard attached to pi. So lets eliminate that possibility.

I am not seeing the Pi connect to my router after i have attempted this. Maybe i am not waiting long enough? how long should this take after the Pi boots up?

I have tried modifying the `octopi-network.txt` as instructed by copying and pasting my SSID and Password.

I have tried only modifying the `octopi-wpa-supplicant.txt` as instructed and still not working. 

What gives? Can you modify the image to have an automatic hotspot like Astroprint to easily connect to wifi? This is so much faster and easier.

I am using the Raspberry PI 2
Edimax wifi dongle
OctoPi 0.14

thanks and sorry this is frustrating!

John-Mc, 2018-02-20:
Can;t help with your main connection issues, but you can find instructions for having the Pi automatically set up a WiFi hotspot, if no known  WiFi networks are in range. Hotspot instructions start about 1/3 of the way down the page.

Cyphen-Cyn, 2018-02-20:
i cant seem to get any connection to work not even a wired one. I would like to see peoples working `octopi-network.txt` and `octopi-wpa-supplicant.txt` to make sure something is not strange in there

Do you have a link for the Hotspot?

John-Mc, 2018-02-20:
Sorry. Left the link to the WiFi hotspot out of my post. I edited to include it now.

John-Mc, 2018-02-20:
You do **not** need to modify Both the octopi-network.txt and the octopi-wpa-supplicant.txt. In fact, I may be mistaken, but I believe having something in the octopi-network.txt interferes with the WiFi Hotspot feature I linked.  I'd comment out any lines you have edited in octopi-network.txt.

Cyphen-Cyn, 2018-02-20:
I am going to start with a fresh image again, just to be safe. Its super fast to re-image anyways. If this does not work ill wait until one of the developers can help and just use Astroprint for the interim.

John-Mc, 2018-02-20:
@Cyphen-Cyn   This page is more for bug reports than for user assistance. You might try the facebook page [OctoPrint Users Group](https://www.facebook.com/groups/piprint/)

Cyphen-Cyn, 2018-02-20:
Seems like a bug, i have done everything it has directed me to do, and it works perfectly with Astroprint. Ill try posting there as well. But this is strange.

John-Mc, 2018-02-20:
@Cyphen-Cyn I updated my post (a couple posts up) with links to other support groups. You probably don't need to post in multiple locations, since many knowledgeable users follow more than one. All of those links are fairly high-traffic sites, so you should get some good responses and troubleshooting assistance. (Please note: I do not have any official capacity on this page, and am not trying to act as the page police. Just trying to point you to other resources in case Guy does not have the time to get back to you here.)

guysoft, 2018-02-20:
Does wired connection work? Just as a test.

Cyphen-Cyn, 2018-02-20:
Wired connection also does not work. I did however get both wired and wireless to work on Astrobox just to make sure I didn’t have a hardware or network issue.

foosel, 2018-02-20:
If even wired doesn't work, a regular Raspbian image most likely won't work either. My money would be on your router not handing out IP addresses via DHCP, which you circumvented through the Astrobox hotspot setup.

> Can you modify the image to have an automatic hotspot like Astroprint to easily connect to wifi? This is so much faster and easier.

If it works. Which it doesn't necessarily with the myriad of possible hardware combinations this image runs into out in the field. This (and licensing issues with regards to hostapd read wifi drivers) is also why I pretty much gave up on bundling netconnectd.

Cyphen-Cyn, 2018-02-20:
Hmm interesting, that makes sense. I will have to see if I can modify my router, however it may be worth looking into the hotspot workflow for OctoPi/Print since there is a plethora of forums and comments repeated to users who are unable to connect.  When I get home from work I’ll give this a try and report back my results. 

Thanks guy and foosel, let me know if you have anything else I can try. Your help is much appreciated!

Cyphen-Cyn, 2018-02-21:
i am still working on this i will post an update when i have something substantial to report, please don't close. :)

Cyphen-Cyn, 2018-02-21:
Okay i have confirmed that DHCP is enabled and working since other computers and laptops that connect via Ethernet are working and receiving an IP address. Any other ideas?

guysoft, 2018-02-21:
mii-tool? Check if there is a link.

guysoft, 2018-02-28:
Closing if no reply, since it seems like the problem is with the route of OP