guysoft/OctoPi #486: Unable to connect to clean install

Arakon, 2018-02-21:
I'm running the latest stable. It apparently successfully expanded the partitions on the SD Card.
I set a manual IP (192.168.0.111).
The Pi starts, LED flickers, the partition is indeed larger. However, I can't connect. Also, it initially assigns the proper IP apparently (looking at my router's list of connected devices), then switches to a DHCP IP suddenly.
FTP, HTTP, SSH all fail to connect with "Connection refused" on either IP. "ssh" file is in the boot partition, but is never deleted. I have no means at all to access the installation.
Note that this was fine with 1.34, until an apparently broken auto-update to 1.35 killed the install completely.

guysoft, 2018-02-21:
How did yyou set tthe manual IP?

Arakon, 2018-02-21:
By editing the octopi-network.txt.
`auto eth0:1
iface eth0:1 inet static
  address 192.168.0.111
  netmask 255.255.255.0
  broadcast 192.168.0.255`

I went back all the way to 2016-03-18-octopi-jessie-lite-0.13.0 and that version works fine, I'm able to connect and print and everything.

guysoft, 2018-02-21:
octopi-network.txt is obsolete and is being removed,  please use the method described here: https://github.com/guysoft/OctoPi/issues/219

Closing because duplicate

closed 2018-02-21 by guysoft