guysoft/OctoPi #490: Logitech QuickCam Messenger not detected by mjpg-streamer

Skystriker, 2018-02-24:
Hi, I got a small issue with the streamer not detecting my webcam.
Sorry again @foosel for posting the bug on the OctoPrint-hub 
and now  stealing your bug template :)

What were you doing?
I shutdown octoprint.
Then I plugged the USB-Webcam in
and restarted Octoprint

What did you expect to happen?
For Octoprint to display an image

What happened instead?
The mjpg-streamer did not start.
The Webcam is seen by the OctoPi via lsusb as Bus 001 Device 005: ID 046d:08f0 Logitech, Inc. QuickCam Messenger

However it is not detected as a Camera as the /var/log/webcamd.log indicates

Starting up webcamDaemon...

Configuration:
camera: auto
usb options: -r 640x480 -f 10
raspi options: -fps 10
http options: -w ./www-octopi -n

No camera detected, trying again in two minutes
No camera detected, trying again in two minutes

I have tried fiddeling with the settings in the octopi.txt, but that didnt solve the problem either.
I have tried editig -r -f and -y.
I noticed that they do not influence the settings in the webcamd.log. Is this normal?
Aditionally I tried to get an image with dwc_otg.speed=1 enabled and disabled in /boot/cmdline.txt.
Now I am out of options.

Did the same happen when running OctoPrint in safe mode?
Yes

Version of OctoPrint
1.3.6

Operating System running OctoPrint
OctoPi

Printer model & used firmware incl. version
Tevo Tornado

Browser and Version of Browser, Operating System running Browser
Chrome Version 63.0.3239.132

Link to octoprint.log
https://pastebin.com/QXkf4fUe

Link to contents of terminal tab or serial.log
https://pastebin.com/5w7WGyW3

Link to contents of Javascript console in the browser
https://pastebin.com/Eg5zEfWk

Screenshot(s)/video(s) showing the problem:
[image]

Thank you for your help.
Sky

guysoft, 2018-02-24:
1. Are you using a power supply with aat least 2A and a GOOD power cable? Its the no 1 miss and due to common cheap power supplies and cables.
2. Can you provide the output of ls /dev/video*
3. Can you provide the output of dmsg?

Skystriker, 2018-02-24:
Thanks for the reply.

Yeah, i am using a 4x2.1A phone charger as power supply.

The output of pi@octopi:~ $ ls /dev/video*
is **/dev/video0**

I guess you mean dmesg?
That would be here https://pastebin.com/LkVJftFP

guysoft, 2018-02-24:
So it finds the camera.

```
Linux video capture interface: v2.00
[    9.511990] gspca_main: v2.14.0 registered
[    9.600294] gspca_main: STV06xx-2.14.0 probing 046d:08f0
[    9.600329] gspca_stv06xx: st6422 sensor detected
[    9.949328] input: STV06xx as /devices/platform/soc/20980000.usb/usb1/1-1/1-1.3/input/input0
```
I have no

Skystriker, 2018-02-24:
Alright. Unplugging did nothing.
However I suspeced a corrupted image and was correct.

So i formatted everything and wrote a new image onto the Pi.
Now the /var/log/webcamd.log displayes this:

logger: Starting USB webcam
Running ./mjpg_streamer -o output_http.so -w ./www-octopi -n -i input_uvc.so -r$
MJPG Streamer Version.: 2.0
 i: Using V4L2 device.: /dev/video0
 i: Desired Resolution: 640 x 480
 i: Frames Per Second.: 10
 i: Format............: JPEG
 i: TV-Norm...........: DEFAULT
 i: The specified resolution is unavailable, using: width 320 height 240 instea$
 i: Could not obtain the requested pixelformat: MJPG , driver gave us: RGB3

Seems like it is seeing the webcam now. But it is not outputting the correct filetype. Is there some way around this?

foosel, 2018-02-24:
Looks like your camera doesn't support mjpg natively. Specifying `-y` in the usb camera options in `octopi.txt` might get it to work, but your Pi will then have to transcode the stream which depending on system resources can negatively affect printing quality. Personally I'd rather switch to a camera that does support mjpg and slightly larger resolutions.

guysoft, 2019-04-23:
Closing due to no reply

closed 2019-04-23 by guysoft