guysoft/OctoPi #494: Stream Jittery

tdub415, 2018-02-28:
I have my octopi.txt settings set as follows:
camera_raspi_options="-x 1280 -y 720 -fps 30 -quality 95 -awb fluorescent"

It seems to be running at/close to 30 FPS but every couple of seconds it jumps/skips several frames, resulting in a very jittery stream.  I recently upgraded to a raspberry pi 3 (was using a zero W) because the zero was maxing out CPU with the above stream settings when the printer was printing circles/round parts.  Trying to narrow down what's causing the stream to jump so much, rather than being a nice smooth video feed.

guysoft, 2018-02-28:
Did you try lowering framerate?
possible duplicate: https://github.com/guysoft/OctoPi/issues/201

tdub415, 2018-02-28:
I started with 10 FPS (which was the best tradeoff for stability I had with the pi zero), and tried 15, 20, 25, and 30, the framerate seemed better, but the stutter/jumps were present even at lower framerates.

guysoft, 2018-02-28:
The webcam image is provided by mjpg-streamer.
So I would check your connection - the Pi3 has a really weak wifi chipset - did you try connecting to ethernet/a usb wifi dongle?

tdub415, 2018-02-28:
ethernet isn't an option where it is set up, so I've only been using the built-in wifi.  I've got an older https://www.canakit.com/raspberry-pi-wifi.html dongle from a pi2 project I think I can locate and try that.  I didn't even think that the wifi chip's strength would potentially be the bottleneck, since I hadn't had any issues with the stream on the zero, other than when streaming and printing the zero was maxing out CPU.

guysoft, 2018-02-28:
I can't debug problems based on what you think. Please try either connecting via ethernet, just as a test, or connecting that wifi dongle. It might also be the camera as explained in #201

tdub415, 2018-02-28:
I will try with the dongle if I can find it once I get home for the day.  If i'm reading #201 correctly, that's with a USB webcam, not the raspberry pi cam, right?

guysoft, 2018-02-28:
Yes, in that issue thee user is using a usb cam.

tdub415, 2018-03-01:
There doesn't seem to be any noticable improvement when using the wifi dongle.  On a whim I tried a different SD card, and that seems to have helped, though it doesn't quite seem like the full 30 fps that I have set

guysoft, 2018-03-01:
You edited out " On a whim I tried a different SD card, and that seems to have helped, though it doesn't quite seem like the full 30 fps that I have set" - found out this is not the case?

This could still be a bandwidth issue, and without putting the Pi close to the router, or using Ethernet, I can't determine if that is the case. Other than that I have no solutions, perhaps lowering the resolution.

tdub415, 2018-03-01:
Yea, trying a different SD card didn't actually help, I had forgotten to update all the settings to match in octopi.txt.  It was seemingly running 'smoother' at 10 fps than 30 fps.  This evening I will see if I have a long enough ethernet cable that I can run to the pi, or if not, temporarily move it and the camera to test.

guysoft, 2018-03-01:
If its running without frame drops on 10fps and not at 30fps,its wither a bandwidth issue or performance issue.