guysoft/OctoPi #495: Octopi won't connect directly with printer

Hedobum, 2018-03-01:
I just ran into this problem after using Octopi for some time now. It's been working great until just recently. Now, when I run Octopi, it doesn't connect with the printer. Com7 is there when I plug the printer into my PC directly, but is gone when I plug into the pi. The pi doesn't register that the printer is plugged in.
Strangely enough, when I connect with Printoid on my phone, I have full access. It connects with Octopi and fires up the printer. I did have to add a username&pass to the profile in Printoid to get it to connect, but that user/pass combo doesn't work when I enter it into Octoprint in my PC browser. It just gives me a "login failed" error. Thing is, I've never needed to enter a user/pass with the PC browser...it just connected automatically as soon as I opened Octoprint. I'm not sure what to do about this now. Forgive me if I sound a bit vacuous...this is driving me nuts that I can't figure this out and it's probably something simple that I'm just overlooking.

guysoft, 2018-03-01:
I am not sure I fully understand.
* Com7 is a windows port numbering and is not relevant to this issue. Linux and the raspberrypi should use a scheme such as ``/dev/ttyUSB0`` or ``/dev/ttyACM0``, depending on the driver
* Printoid is an interface which directly pulls information from OctoPrint's interface, it does not invent its own thing - are you sure you are connecting to the same OctoPrint instance? If the password does not work and the printer is not seen it sounds like two separate installations.

Hedobum, 2018-03-01:
It is exactly as I  said...the printer shows up when I connect it directly through my pc (obviously). It does not show up when I connect with Octopi  (USB plugged directly to the pi) but when I use this third party app, it connects and controls the printer. When I control through this app, I can see all the functions being executed in the browser on my pc but I  have no control from that end.
There's no other instance of Octopi. Both the app and my browser are connecting to the same address and I only installed Octoprint once on the pi.

guysoft, 2018-03-01:
Are you logged in from your PC browser?

John-Mc, 2018-03-01:
It sounds as though for whatever reason, OctoPrint is not recognizing the login from his PC browser. I'm not sure why this would be, unless his username or password has been changed or is being entered incorrectly. I did read somewhere on one of the OctoPrint/OctoPi web sites of how to reset the login (basically causing it to run the set-up wizard again next time you call up OctoPrint in your web browser). I wonder if that would help here?

Hedobum, 2018-03-01:
I am logged in through the browser (or so it would seem). Octoprint is open in the browser...it just doesn't connect to the printer from the browser. When I connect with the phone app and load up the printer, I can see it loading with the browser and I can watch the printer being controlled but I have no control of any functions from the browser.

guysoft, 2018-03-01:
Try a different browser on the PC perhaps. Might be a caching iasue. Anyway, an Octoaprint issue, not an OctoPi issue.

Hedobum, 2018-03-02:
Ok....I thought they were basically the same thing. 
Regardless, I think I have to start from scratch. Easier than trying to figure out why it's not working properly...just a pain in the south end.