guysoft/OctoPi #496: Can not connect to CH340G usb to serial with octopi image

Custom3DProducts, 2018-03-02:
The standard octopi image does not include the drivers needed to connect to the CH340G usb to serial chip. 
This makes some boards and ramps configurations unusable.
know printer board with CH340G is MKS GEN L

guysoft, 2018-03-02:
strange,  I have a printer with MKS gen L and it works fine with Marlin.
What firmware? What OctoPi version?

trajesus, 2018-03-03:
My anet a8 board uses the CH340G chip and my octopi can connect perfectly. However, not in the auto mode, I had to specify the baud rate and the serial port after that it's been running perfectly for almost a year :)

guysoft, 2018-03-03:
Auto mode never worked for me, I think its dead,  @foosel?

foosel, 2018-03-05:
No, it's not "dead". Port detection works, but it depends on there either being only one port or the printer controller to allow entering programming mode. And baudrate detection depends on the printer actually replying to probes in a timely fashion.

guysoft, 2018-03-05:
@trajesus What is the output of lsusb?

trajesus, 2018-03-05:
I don't have any problem with my octopi, but if you want to know the output when I arrive home I'll post it here :)

guysoft, 2018-03-05:
Ah, then nevermind